guysoft/CustomPiOS #1: Better build tracking in stdout

guysoft, 2017-07-16:
Currently strout is a total mess out bash command output, which is great for ``build.log``, but it should really display something like>

building module<x>

Perhaps also add debug to only print certain modules. Out, and print the last lines of build.log if the build fails.
Also alert of a module is broken.

jeffeb3, 2018-06-04:
The `set -x` is awesome for the build.log though. Not having to try to add trace statements is really really helpful. I guess you'd have to redirect stdout to build.log, and have another logging mechanism that went to the console. I can think of how to do this in python, or C++, but I don't know how you'd do that in bash.

The output I really like is from catkin-tools. They build each package using catkin, and store the console output in files, and then if there's a warning or an error, they replay parts of that build file. The output is a lot of headers about which components are building (it's a multithreaded build) and how long they are taking. If there are any warnings or failed packages, those parts get printed (with colors!).

If the modules were really the main source of errors, then I could see using set -x only in the module's scripts, and forwarding those to output log files (more files are cheap, modules have to have unique names anyway). On an error, you could either parse the build file (if you wanted to be cool) or just cat it to the console, or even just point the user to the file (although that's a bit annoying for the docker containers, since the file gets deleted after it builds, but that might be a different issue).

Problems in custompios would need to be handled differently. Especially things like qemu or chroot not working.

guysoft, 2018-07-10:
Yes, 
Something that takes stdin, logs it, and prints out more info could be useful. Perhaps sending signals to it for "message information" could work, and only print the last few commands on fail. 
Python3 is already a dependency, so I will be fine with this component was written in python3.
For a start it could go here: https://github.com/guysoft/CustomPiOS/blob/devel/src/build#L30
There is a tee command here which could be replaced with something a little smarter.

OutsourcedGuru, 2018-12-10:
I tend to wrap my included fork/spawn output into a log by sandwiching it in something like:

```
--- ls -l go ----------------------------------------------
total 0
drwxr-xr-x  7 justincase  staff  224 Aug 31 20:25 bin
drwxr-xr-x  3 justincase  staff   96 Dec  4  2017 pkg
drwxr-xr-x  8 justincase  staff  256 Aug 31 20:35 src
-----------------------------------------------------------
```