guysoft/CustomPiOS #4: Doesn't seem to work with Stretch

tenten8401, 2017-09-04:
Fired this up in a clean install of Ubuntu 16.04 with a Raspbian Stretch image. https://www.raspberrypi.org/downloads/raspbian/

Specifically Raspbian Stretch With Desktop.
It just seems to hang at a blinking cursor when booting.

guysoft, 2017-09-10:
What distro are you building? The default example?

tenten8401, 2017-09-11:
@guysoft I'm building the default example, no gui module.

guysoft, 2017-09-17:
Works for me, can't reproduce.

Can you provide the build output?
Also attaching the scripts used to build so you can compare, includes the build.log.
[ExampleOS.tar.gz](https://github.com/guysoft/CustomPiOS/files/1309052/ExampleOS.tar.gz):

Proof of it running here
[image]

tenten8401, 2017-09-17:
Well that's baffling. Probably something I did on my end then.
Either way, I think I've got another solution in place for building images.

closed 2017-09-17 by tenten8401

guysoft, 2017-09-17:
What are you using,  if I may ask?

tenten8401, 2017-09-17:
Just cloned the repo, ran through this: https://github.com/guysoft/CustomPiOS#build-a-distro-from-within-raspbian--debian--ubuntu--custompios-distros, dropped the image in there and built it. Used a Ubuntu 16.04 VirtualBox VM.

guysoft, 2017-09-17:
Did you run ``make_custom_pi_os``?
I think you missed that step and built the template.

Also I was asking what were you using as an alternative?

tenten8401, 2017-09-17:
I ran make_custom_pi_os and everything in that section I linked you, putting the stretch desktop image in before finnaly building the final image.

As for the alternative, I'm using https://github.com/RPi-Distro/pi-gen, which takes about 45 minutes to build on my system with apt caching.