guysoft/CustomPiOS #9: Variables are not escaped properly

TheAssassin, 2017-10-13:
While creating #8, I fixed some of the variable occurrences with improper escaping. The scripts should be revised, and the variables should be escaped appropriately. This is probably quite some work, but should prevent the usual annoyances (like "there must be no space in the paths" etc.).

guysoft, 2017-10-13:
Would appreciate it if anyone wants to take this endeavor. It sorta works for me and there might be there someone who is better in Bash coding conventions. @shlomif I remember you were pretty good at it. If you have any pointers that would be appropriated.

shlomif, 2017-10-14:
@guysoft : see http://shlomif-tech.livejournal.com/14671.html and http://perl-begin.org/topics/security/code-markup-injection/ .