guysoft/CustomPiOS #15: Doubt!

philo-math, 2017-12-25:
I am unable to understand the purpose of this project since readme.rst is not much informed.

My queries:
- Is this project works on developing custom BIOS for rpi.
- Or just port other OS for rpi.

Pardon me for the silly doubt but it will help me to understand its purpose.

guysoft, 2017-12-25:
The purpose of this project is to create a custom distribution for the Pi and Armbian devices.
It takes a raspbian/armbian image, opens it, modifies it and closes dor shipping. You can see a list of based distroa in the readme.

The idea that its much faster and easy to maintain than debootdtrap and supports modules across distros.

guysoft, 2017-12-26:
[Added to FAQ](https://github.com/guysoft/CustomPiOS/wiki/FAQ)

closed 2017-12-26 by guysoft