guysoft/CustomPiOS #18: When a module fails to build, CustomPiOS keeps building the other modules

guysoft, 2018-02-18:
A failure should halt the build, the problem happens because there is no testing in ``/distro/src/workspace/chroot_script`` that is generated by [execution_order.py](https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/execution_order.py).

The code there should be improved.

foosel, 2018-02-21:
Just ran into this as well. Was confused why my image wasn't working, then noticed that the OctoPi build had run into a network error half way through but no error was generated :/

guysoft, 2018-02-21:
Yes, also @toddejohnson ran in to it. I was going to do this, even added [echo_red](https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/common.sh#L24) and [echo_green](https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/common.sh#L28) so we could actually print this in color, but never managed to capture the failures. If anyone has time to do this then awesome, if not I will get to it eventually.

commit a5ea5201 (2018-02-21): Merge pull request #21 from foosel/fix/error_handling Fixes #18

Properly detect build errors
closed 2018-02-21 by guysoft