guysoft/CustomPiOS #22: Building on windows

guysoft, 2018-02-25:
Hey,
So this weekend I was playing around with building OctoPi on windows (and hopefully after that other CustomPiOS distros).

My current work is available on the [windows  branch on OctoPi](https://github.com/guysoft/OctoPi/tree/windows)
and [windows branch on CustomPiOS](https://github.com/guysoft/CustomPiOS/tree/windows).

I managed to run a successful build and I will explain how and the limitations.

Before I start I'll mention - no native build is planned, only using the vagrant method, because its the only way.

**Requirements**
  1. Vagrant
  2. Virtualbox
  3. Power shell or bash (see decision)

**limitation (why does't windows work now)**
The main limitation in supporting windows is no way to preform loopback mounts on synced folders in windows.
CustomPiOS mounts the img file form the workspace, and the workspace sits in a synced folder. This is supported by NFS, but not by any other method. And NFS is not supported by windows. I also tried the [winnfsd plugin](https://github.com/winnfsd/vagrant-winnfsd). But it was unstable, sometime it writes, sometimes it skips. Which means things are random and usually the build fails.

The workaround is simply - [build inside the VM](https://github.com/guysoft/OctoPi/blob/windows/src/config#L6) and them move the image outside to the syned folders when its done.
So I wrote a [small script to set the CustomPiOS path](https://github.com/guysoft/CustomPiOS/blob/windows/src/update-custompios-paths.ps1). and a [vagrant-windows build](https://github.com/guysoft/OctoPi/tree/windows/src/vagrant-win). If you run those too, similar to their linux/mac equivalent, then a successful build will go. The [release script](https://github.com/guysoft/CustomPiOS/blob/windows/src/release) needs to be updated still.

**Decision that need to be made**
I would prefer to maintain everything using bash, but bash needs to be installed and I am not sure if windows users tend to download [git for windows](https://git-scm.com/download/win) which I've been using. So the question is do I make it a requirement, or implement everything twice with powershell.

For now, here is picture of a successful build in windows:
[image]

foosel, 2018-02-25:
Personally I don't use anything *but* the bash on Windows anyhow (regular CMD is too limited and I never got much into power shell). And in the interest of better maintenance I think it's fine to just add some working bash on the path to the set of requirements. I guess it's not many people building this on Windows anyhow (I would probably be one of them though), and having to install a bash is way better than having to manually install a guest OS on a VM which is the approach now.

guysoft, 2018-02-25:
Just to be clear there will still be a VM (at the moment debian/stretch) with a gust OS. Because AFAIK windows does not support loopback mounting and chroot. Also the VM is there to keep the environment the same across systems. I even use it within my Ubuntu.

foosel, 2018-02-25:
I know. The keyword here was "manually install".

guysoft, 2018-02-27:
Ok, so when I'll have time I'll try it out with bash and set instructions.
Another thing I think this will require is that ``src/vagrant`` folder would be managed by ``CustomPiOSupdate-custompios-paths``, because there would be more logic there.

guysoft, 2018-12-23:
I think I am closing this, and directing people on windows to build via docker. Would appreciate it if someone reports that indeed building from docker on windows works.

guysoft, 2019-01-24:
Yep, docker is released on 1.1.0. And any windows build will involve some form of virtualization of a linux kernel.
I have no windows machine and time to test this, would really value it if someone writes up a guide. But for now closing this. Feel free to comment here if you have any windows-building issues. We can always reopen.

closed 2019-01-24 by guysoft