guysoft/CustomPiOS #23: aarch64  OrangePi Zero Plus

mr-miky, 2018-03-06:
'm trying to generate an image to useon  Orange PiZero Plus (H5 4 core 64bit) armbian variant
I'm using the OPi0 + to build the image, it seems to be all successful but then the script tries to run /bin/bash with Qemu-arm-static generates an error because it tries to run bash aarch64 with qemo-arm (32bit) .

You probably need to make sure that the right qemu is used according to the target architecture ....

guysoft, 2018-03-06:
Can you provide a download link to the image you are using as a base image?

mr-miky, 2018-03-06:
but fails later ....

+ mkdir -p /var/cache/apt/archives
+ apt-get clean
+ apt-get update
Get:1 http://security.debian.org stretch/updates InRelease [63.0 kB]
....
Get:135 http://cdn-fastly.deb.debian.org/debian stretch-backports/main armhf Packages [267 kB]
Get:136 http://cdn-fastly.deb.debian.org/debian stretch-backports/main armhf Contents (deb) [3068 kB]
Get:123 http://cdn-fastly.deb.debian.org/debian stretch-backports/contrib Translation-en 2018-02-21-1428.52.pdiff [1041 B]
Get:132 http://cdn-fastly.deb.debian.org/debian stretch-backports/non-free arm64 Contents (deb) 2018-02-21-1428.52.pdiff [497 B]
Reading package lists...
E: Could not open file /var/lib/apt/lists/httpredir.debian.org_debian_dists_stretch-backports_main_binary-armhf_Packages .diff_Index - open (2: No such file or directory)
E: Could not open file /var/lib/apt/lists/httpredir.debian.org_debian_dists_stretch-backports_main_Contents-armhf.diff_Index - open (2: No such file or directory)
+ exit 1

Why does the script require armhf files ??

mr-miky, 2018-03-07:
Workaround is to add
rm -rf /var/lib/apt/lists/*
in
CustomPiOS/src/variants/armbian/pre_chroot_script 

```
#cleanup
rm -rf /var/lib/apt/lists/*
mkdir -p /var/cache/apt/archives
apt-get clean
apt-get update
```
and in

OctoPi/src/modules/octopi/start_chroot_script

```
and unpack /filesystem/boot /boot
rm -rf /var/lib/apt/lists/*
apt-get update
```

guysoft, 2018-03-07:
Strange,  perhaps you have out of date files in the ``workspace`` folder?

mr-miky, 2018-03-08:
No, they seem to be already present in the image...
https://dl.armbian.com/orangepizeroplus/Debian_stretch_next.7z

guysoft, 2018-03-08:
See answer in #25 ,  touch the file, I suggest you make a new variant and we can add it here too.

guysoft, 2018-04-01:
Any news? Should I close this?

mr-miky, 2018-04-05:
I practically could not complete the image generation. I would like to finish the thing but unfortunately I do not have the free time to invest in this project.
Close the report if you think it's appropriate.

guysoft, 2020-01-07:
Hey, updating I stared looking on building aarch64bit distros. At the moment you need a device that can execute both aarch64 and armf in many cases, which means you need a Rpi or an equivelent. The docker build method seems to work.

guysoft, 2020-01-16:
Fixed, you can now set in your config:
```
export BASE_ARCH=aarch64
```
And then the 64bit qemu version would be used.

closed 2020-01-16 by guysoft