guysoft/CustomPiOS #24: Figure how to release versions

guysoft, 2018-03-06:
Hey,
So at the moment CustomPiOS is picking up momentum, and other people are using it too. And its time to start tagging releases, so people know what they built against when releasing distros.
The question is - how should versions be organized?
I could just tag when I feel like it, but would like to hear when there is any way other people would prefer?

Once this happens CustomPiOS will have a master branch with the versions tagged, and a devel branch which would be the one updating regularly.

guysoft, 2018-03-06:
cc @foosel @toddejohnson

foosel, 2018-03-09:
To be honest, I don't think I have a preference, other than being consistent ;) In general I try to stick to [semantic versioning](https://semver.org/) for my own projects.

guysoft, 2018-03-09:
Ok, so that would make the major go up fast. CustomPiOS 4.0 in a year
probably.

We could do that.
I think of starting from 1.0.0.

foosel, 2018-03-09:
Are you planning on making "incompatible API changes" all the time 🤨? Because that is the only case that should make major increment.

guysoft, 2018-03-09:
Come to think of it, there is a lot of enhancement when I change stuff.  But no breaks.

foosel, 2018-03-09:
So you'd be at 1.23.5 in a couple of months. No big deal ;)

toddejohnson, 2018-03-09:
I agree! Sadly if you branch for devel I'll probably just chase master for now as checking out tags is a pain and I'm lazy.

guysoft, 2018-05-29:
Done

closed 2018-05-29 by guysoft