guysoft/CustomPiOS #25: Wrong size expansion

mr-miky, 2018-03-07:
I do not know if it's a mistake or I've lost something.
I'm trying to generate an image that works on the orange pi zero plus.
I downloaded the specific armbian image for OpiZero plus. I did all the required steps and launched sudo bash -x ./build_dist armbian.
When the image is resized, something strange happens: the resize is always + 400Mbyte even if / CustomPiOS / src / variants / armbian / config explicitly reports BASE_IMAGE_RESIZEROOT = 500

+++ echo 'Resized parition 1 of Armbian_5.38_Orangepizeroplus_Debian_stretch_next_4.14.14.img to +400 MB'

I tried other values but the resize is always 400. The only solution I found was to modify CustomPiOS / src / common.sh function enlarge_ext () changing size = $ 3 in size = 750

guysoft, 2018-03-07:
That's because the armbian variant sets its own value of ``500`.
See https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/variants/armbian/config

In order to override this you can create a ``config.local`` in the OctoPi folder and input in it a different value.

guysoft, 2018-03-08:
Ok, can you provide ``build.log``?

guysoft, 2018-03-08:
Your build fails and does not even reach using ``BASE_IMAGE_RESIZEROOT``.
Perhaps you are mixing between ``BASE_IMAGE_RESIZEROOT`` and ``$BASE_IMAGE_ENLARGEROOT``?

* ``BASE_IMAGE_ENLARGEROOT`` resizes the image and gives you space to build. 
* ``BASE_IMAGE_RESIZEROOT`` minimizes the size of the image when you are done at the end, giving a minimal image file.

you could add to the armbian variant ``touch /lib/systemd/system/regenerate_ssh_host_keys.service`` to fix your current fail. Or better yet, copy the folder a side to armbian-orange-pi-zero-plus and build that, (write armbian-orange-pi-zero-plus instead of armbian). That way we can merge it back in to CustomPiOS when you are done for everyone.

mr-miky, 2018-03-09:
I understood that BASE_IMAGE_RESIZEROOT is to enlarge the partition so as to be able to insert octoprint, Cura etc.
How can I expand more than 400Mbyte and put slic3r and more in it?

So if I have not misunderstood to add space to the partition I have to declare
BASE_IMAGE_ENLARGEROOT = 750
 in CustomPiOS/src/variants/armbian/config  ?

I have disabled the compilation of Cura engine because among other things g ++ and gcc are available up to a maximum of 4.7. The default gcc is gcc6.
I do not know if Cura engine can be compiled correctly with gcc 4.7 or maybe with gcc 6.

This night I try with the touch ... and see if at least the construction of the image is completed

guysoft, 2018-03-09:
Yes, use BASE_IMAGE_ENLARGEROOT = 750

mr-miky, 2018-03-09:
None , new compilation errors :-(

guysoft, 2018-03-10:
Please post build.log when you get errors. 
I can't help without the log.

mr-miky, 2018-03-10:
I know but I was too tired and I went to bed.

```
removed 'filesystem/root/etc/haproxy/errors/503-no-octoprint.http'
removed directory 'filesystem/root/etc/haproxy/errors'
removed 'filesystem/root/etc/haproxy/haproxy.cfg'
removed directory 'filesystem/root/etc/haproxy'
removed directory 'filesystem/root/etc'
removed directory 'filesystem/root'
removed directory 'filesystem'
+ execute_chroot_script /home/pi/CustomPiOS/src/modules/disable-services /home/pi/CustomPiOS/src/modules/disable-services/end_chroot_script
+ cp -vr --preserve=mode,timestamps /home/pi/CustomPiOS/src/modules/disable-services/filesystem .
cp: cannot stat '/home/pi/CustomPiOS/src/modules/disable-services/filesystem': No such file or directory
+ true
++ which qemu-arm-static
+ cp /usr/bin/qemu-arm-static usr/bin
+ cp /home/pi/CustomPiOS/src/modules/disable-services/end_chroot_script chroot_script
+ chmod 755 chroot_script
+ cp /home/pi/CustomPiOS/src/common.sh common.sh
+ chmod 755 common.sh
+ chroot . usr/bin/qemu-arm-static /bin/bash /chroot_script
+ rm -r /usr/sbin/policy-rc.d
+ rm chroot_script
+ rm -rfv filesystem
+ execute_chroot_script /home/pi/CustomPiOS/src/modules/password-for-sudo /home/pi/CustomPiOS/src/modules/password-for-sudo/start_chroot_script
+ cp -vr --preserve=mode,timestamps /home/pi/CustomPiOS/src/modules/password-for-sudo/filesystem .
cp: cannot stat '/home/pi/CustomPiOS/src/modules/password-for-sudo/filesystem': No such file or directory
+ true
++ which qemu-arm-static
+ cp /usr/bin/qemu-arm-static usr/bin
+ cp /home/pi/CustomPiOS/src/modules/password-for-sudo/start_chroot_script chroot_script
+ chmod 755 chroot_script
+ cp /home/pi/CustomPiOS/src/common.sh common.sh

```

first it removes the filesystem directory and then gets angry because it does not find it. :-)

mr-miky, 2018-03-10:
Evidently I did not sleep enough because I did not realize that the problem is out of the chroot

guysoft, 2018-03-10:
Ok, does that mean you managed to solve the size issue and I can close this issue?

closed 2018-03-11 by mr-miky