guysoft/CustomPiOS #29: qemu-arm on an arch-linux based laptop.

jeffeb3, 2018-04-30:
Hey Guy,

I switched to using an arch derivative for my home computer, and the "native" version isn't building a custom Pi os.

For starters, the qemu-arm-static is named "qemu-arm" (and I know nothing about qemu, so I'm just thinking it's the same because it prints the same help message :D ). I switched the custompios script to look for that, but it's erroring with:

`chroot: failed to run command 'usr/bin/qemu-arm': No such file or directory`

So it seems like I'm not doing that right, but in fact, it is there. I added a few prints to the custompios script:

```++ which qemu-arm
+ cp /usr/bin/qemu-arm usr/bin
+ cp /home/jeff/src/CustomPiOS/src/modules/base/start_chroot_script chroot_script
+ chmod 755 chroot_script
+ cp /home/jeff/src/CustomPiOS/src/common.sh common.sh
+ chmod 755 common.sh
+ pwd
/home/jeff/src/test/src/workspace/mount
+ ls -al usr/bin/qemu-arm
-rwxr-xr-x 1 root root 4557848 Apr 30 11:57 usr/bin/qemu-arm
+ ls -al bin/bash
-rwxr-xr-x 1 root root 912712 May 15  2017 bin/bash
+ ls -al chroot_script
-rwxr-xr-x 1 root root 1331 Apr 30 11:57 chroot_script
+ chroot . usr/bin/qemu-arm /bin/bash /chroot_script
chroot: failed to run command 'usr/bin/qemu-arm': No such file or directory
```

So the qemu-arm is found, copied into the mount/usr/bin folder, and chroot . usr/bin/qemu-arm says "No such file or directory"... Weird. Maybe it's something wrong with chroot?

I know arch is a rolling release, so the versions are often newer. The chroot version is 8.29 and qemu-arm is 2.11. BTW, I installed qemu using `qemu-headless-arch-extra 2.11.1-2`.

Here is the "full" log:
[build.log](https://github.com/guysoft/CustomPiOS/files/1961657/build.log)

I'm hoping I don't have to use vagrant on a brand new Linux OS, but if I have to, I guess I'll try that next.

guysoft, 2018-04-30:
Use the vagrant build method.

jeffeb3, 2018-05-28:
The vagrant build was giving me some trouble (the MFS mounts and openbox weren't agreeing with me). I instead set up a Dockerfile to do it. Here is a diff of what I did:

https://github.com/jeffeb3/v1pi/compare/master...docker_build

guysoft, 2018-05-29:
1. Cool! Thanks, I was hoping someone would do this. Should I pull this to CustomPiOS so we have a new official build method?
2. Do you want to add this distro [here](https://github.com/guysoft/CustomPiOS#list-of-distributions-using-custompios)? 
3. I see you also need the net modules feature so you don't need to symlink to OctoPi's module. Ill push it up my list.

jeffeb3, 2018-05-29:
https://github.com/guysoft/CustomPiOS/pull/30

closed 2018-11-24 by jeffeb3