guysoft/CustomPiOS #35: Allow to choose custom "config" file when building

MaitreDede, 2018-09-18:
Hi,
I am integrating the build process into a gitlab CI,and it is possible to define the "config" file name before calling the `build_custom_os` script, so I can create one config file per target I would like to build.
Thanks.

guysoft, 2018-09-19:
Also - if you have CI solutions that work for you sharing would be great to help others. Due to loopback devices it can be tricky (the reason I didn't get docker working yet).

MaitreDede, 2018-09-26:
I will try and give feedback... If I remeber, I managed to achieve CI building of raspbian image...
I will try to get the code, and give feedback :)

guysoft, 2018-12-08:
1. any news?
2. There is a docker build you might like now, instruction in the wiki

MaitreDede, 2018-12-09:
Hi,
I saw the new docker build, I love it. But I can't build a CustomPiOS yet.
My env is Gitlab, and the first problem I have is docker : `cannot enable tty mode on non tty input`. I will try to fork and do some tests (the first will be removing the `t` from `docker run -it`).

On local computer, I first build you docker image : I had an issue with the initial `build_docker` file, but I ran docker build myself.
Then I made a git repo with CustomPiOS as submodule, a "prepare" task that download the raspbian distro (if not in cache). And I have 2 operating system in this repo. They don't share modules (yet), but they build fine.

guysoft, 2018-12-09:
Yes, removing the ``-T`` should work. Also if its automated the ``-i`` for interactive should go too.

closed 2019-03-07 by guysoft