guysoft/CustomPiOS #36: Option to shrink result image

MaitreDede, 2018-09-26:
Enlarging the image can be usefull to install apps, but reducing the result image can be also interesting.

For example, you can expand image by 4Gb, do the stuff, and then use a tool like https://github.com/Drewsif/PiShrink to reclaim unused space left, to have a minimal size image (faster to write on SD) that will auto-expand on first boot.

Maybe it can be an option in the config file

guysoft, 2018-09-27:
This is already an option in the base module. See ``BASE_IMAGE_RESIZEROOT``
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/base/config#L26
Note the way its calculated is image size + x extra space.

You can also set that you want the result image to have a small ``BASE_IMAGE_RESIZEROOT`` , but a large building space with ``BASE_IMAGE_ENLARGEROOT``:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/base/config#L23

We really need this info in the wiki, if anyone wants to write a "module settings" entry.

guysoft, 2018-11-20:
Closing if no reply