guysoft/CustomPiOS #38: Octopi overwriting variant files on first real boot

pantsmanuk, 2018-12-14:
Hi Guy/Gina

I'm attempting to turn Prusa 3Ds Octopi fork into a variant (haven't put it onto Github yet, can do if it'll help) so that it can be applied more easily to "central code" - their changes (on the face of it) are configuration for the most part, and it all goes swimmingly right up until I burn the image to SD and boot off it.

The majority of the variant is a config.yaml and printer profile (_default.profile if I'm being pedantic about replicating the fork). I've used post_chroot_script to apply the filesystem changes, and when I qemu_boot.sh the image, everything is right, the config.yaml is there, the printer profile is there with it's changes.

As soon as I boot that same image burned to an SD card, it looks for all the world like both config.yaml and _default.profile are overwritten (entirely in the case of _default.profile, and partially in the case of config.yaml)?

guysoft, 2018-12-16:
Closing if no reply

pantsmanuk, 2018-12-16:
I don't have a copy of the code at home at present. Will upload tommorrow when I'm back at work (GMT timezone)

pantsmanuk, 2018-12-17:
https://github.com/pantsmanuk/PrusaPi

guysoft, 2018-12-17:
Ok I got you - could it be that your SD is faulty? When an SD card dies it will report that it wrote, but wont actually change the files on disk.
Generally it looks fine, and if you see on qemu boot the right files it should work.

And just to be sure:
@foosel Does octoprint on boot overwrite /home/pi/.octoprint/config.yaml?
Perhaps you need to mark something as not first boot?

foosel, 2018-12-17:
> Does octoprint on boot overwrite /home/pi/.octoprint/config.yaml?

No. And otherwise no existing OctoPi installation out there would have the webcam, cura engine, discovery or anything else predefined in that configured, which they do have ;)

guysoft, 2018-12-17:
Ok, so betting its a faulty SD card.

pantsmanuk, 2018-12-17:
Just to try and help myself as much as you, the changes in config.yaml aren't a problem: it's "brand new settings" and re-arranging the file I'm starting from (so, long story short, ignore mention of config.yaml going forward).

The problem is that _default.profile is being "nuked". Screenshot from a diff tool attached: left is "my file", right is what is on the SD card once it finishes boot/expand/reboot.

[image]

foosel, 2018-12-17:
Does `octoprint.log` contain anything after having found an invalid default printer profile after the initial boot? Because that looks like this situation, in which case it will recreate the original default profile.

pantsmanuk, 2018-12-17:
Thanks @foosel. I blew away the SD and tested it with h2testw, but that only confirmed that the capacity is correct (Sandisk card purchased from a reputable vendor, I expected the capacity to be correct), so I'll reflash it and check. Just to be 100% on this, would the "id:" not matching the filename/config.yaml value cause this?

pantsmanuk, 2018-12-17:
```
2018-11-13 14:-2:49,264 - octoprint.printer.profile - WARNING - Invalid profile: /home/pi/.octoprint/printerProfiles/_default.profile
```

OK, so I've gotten something wrong and it's being replaced as a result. @foosel has pointed me in the right direction, I'll take back ownership (in shame).

closed 2018-12-17 by pantsmanuk

guysoft, 2018-12-17:
There is no shame in finding a bug and fixing it.  If you write something
and it works the first time then don't trust it.