guysoft/CustomPiOS #39: [ReqForInfo] Adding a suffix to DIST_VERSION in variant file?

pantsmanuk, 2018-12-18:
Is it possible to either add a suffix to DIST_VERSION, or over-write DIST_VERSION completely (while referencing the module value) in a variant file (guessing config, but any of X_chroot_script would suffice)?

I've tried:

'''
DIST_VERSION+="PE"
'''

and 

'''
DIST_VERSION=$DIST_VERSION+"PE"
'''

in src/variant/foo/config, but neither of them changed the image DIST_VERSION ("0.16.0").

Prusa are adding "PE" to the DIST_VERSION and it'd be nice to replicate that in my variant.

guysoft, 2018-12-19:
Ok, there is a cascade issue here.

Originally the variant system is used to manage different devices or variants of the same distribution. However you want a distro which has a separate version from the original, which is not something this was originally planned for.

The current sourcing hierarchy looks like this:
variant (CustomPiOS folder)
config.local (In CustomPiOS folder)
config (In distro folder)
config.local (In distro folder)
modules config files

or:
[image]|

Its defined in this file:
https://github.com/guysoft/CustomPiOS/blob/devel/src/config

So anything you write in the variant would get overwritten by config file in the distro folder. 
You could technically override it using either a ``config.local`` file in the distro folder as a workaround.

Alternatively we should figure out if  there is a place to extend the way CustomPiOS sources files to support what you are doing.

guysoft, 2018-12-19:
Sure, if you want, we could later on come up with a real solution.  For
example via a better module system or by implementing a "flavor" instead of
a "variant". Which is something @mrbeam originally started adding in to the
variant system. PRs are welcome.

guysoft, 2018-12-19:
Should I close this for now? Or do you have more input?