guysoft/CustomPiOS #45: [feature] set name of img on release

guysoft, 2019-03-07:
See:
https://github.com/zynthian/zynthian-sys/issues/66#issuecomment-468244913

tl;dr
CustomPiOS should have a variable that lets you set the file name of the image.
Should be used here:
https://github.com/guysoft/CustomPiOS/blob/devel/src/release#L16

cc @jofemodo

WheresWaldo, 2019-05-29:
I agree with this or at the very least include the versioning used in Distro-name/src/config to add the versioning number to the build process

guysoft, 2019-06-05:
@WheresWaldo That happens in the base module:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/base/start_chroot_script#L55

WheresWaldo, 2019-06-05:
Maybe I am not understanding how to actually do it. Or maybe this wasn't the correct feature request. What I want is the final image created not to be called YYYY-MM-DD-raspbian-stretch-lite.img but rather by the DIST_NAME-DIST_VERSION.img

guysoft, 2019-06-05:
@WheresWaldo Yes, this is why this issue is open, note I wrote where it needs to be changed. ``$FILENAME`` should have more options.

WheresWaldo, 2019-06-05:
Sorry for being so dense, but I have one more question with regard this option,

In my distro, I am building everything in base( ) under export MODULES=, would I just add release to my list of modules? Such as this:

export MODULES="base(list of modules),release" or would it be
export MODULES="base(list of modules,release)"

guysoft, 2020-01-14:
@WheresWaldo release should not part the modules IMHO, because it does not alter the actual data inside the image.
I could add some setting such as  ``RELEASE_ON_BUILD=true``

guysoft, 2020-01-14:
Ok, no document in the wiki yet, but added two new variables you can set in config:

```
BASE_RELEASE_IMG_NAME=default
BASE_RELEASE_ZIP_NAME=default
```

If you change them to something other than default it should rename the image. I haven't fully tested it so please report if you have any issues @WheresWaldo @jofemodo

commit 4794c862 (2020-01-14): Handle ubuntu release file name and also fix #45
closed 2020-01-14 by guysoft