guysoft/CustomPiOS #46: Loop doesn't seem to be working with Ubuntu 19.04, fails at losetup

WheresWaldo, 2019-04-21:
Building my own version of OctoPrint with a few plugins when the build command is issued and it appears to go all the way through building the necessary modules and ripping apart the image, only to fail rebuilding the image.

Using Ubuntu 19.04 (64-bit) in a VirtualBox VM from osboxes.org. Simply using 'terminal' and not trying to install in Docker or with Vagrant.

[build.log](https://github.com/guysoft/CustomPiOS/files/3101230/build.log)

I am pretty sure it is a permissions issue, but can't figure out how to fix it, sorry I am a linux noob.

A couple of notes, 'realpath' is not in the current Ubuntu package repository so it cannot be installed, also 'curl' needed be added to the install list in order to download the source via the provided instructions. Also, loop is not built into the kernel and not a loadable module so 'modprobe loop' does nothing. Nor does it list loop when searching for modules with 'lsmod'.

I studied the four other issues that mention loop failing (2 closed, 2 open) and it does not appear to provide a clear understanding of what is going on.

Just to verify if my setup of OctoPrint was the culprit, I tried building AlarmPi and OctoPi and both failed at the same location.

guysoft, 2019-04-21:
Ok, tested here on my 19.04 installation, and it works.
In your run it says:
```
failed to setup loop device for
```

Are you sure the loopback device is mounted? Did you use:

sudo modprobe loop

Note - if you did something to your system, I can't really support it, that
is why we have vagrant and docker, so you don't get environment issues.

WheresWaldo, 2019-04-21:
Loop is not a loadable module in Ubuntu, it is no compiled into the kernel, so::

sudo modprobe loop

does nothing

WheresWaldo, 2019-04-21:
I have since created a new VM using Debian 9 and it is appearing to work. I still want to get it installed on Ubuntu since I am more familiar with it than Debian. For now I am going to create a new Ubuntu VM and see if it isn't osboxes.org nonsense. Please leave this open until I report back that it is working.

WheresWaldo, 2019-04-21:
Okay, I can now say with confidence that osboxes.org prebuilt VM for Ubuntu is the fault. I created a new Ubuntu VM installed VBoxGuestAdditions and was able to successfully build a rPi image. please note there are two corrections to your instructions needed if running on Ubuntu 18.04 RTS or newer (I ran CustomPiOS on Ubuntu 19.04). the first command to install dependencies needs realpath removed.

    sudo apt-get install gawk util-linux qemu-user-static git p7zip-full python3

Secondly, since loop is now built into the kernel, there is no need to execute the following command:

    sudo modprobe loop

I am closing this issue as it appears it was a configuration issue with the pre-built VM.

guysoft, 2019-05-06:
Ok, I found out why this happends.

When running losetup you get a list, including

```
/dev/loop7
               0 50331648         0  0 /distro/workspace/2019-04-08-raspbian-stretch-lite.img (deleted)

```

If you run ``losetup -d /dev/loop7`` it solves the issue.
Will try write up a fix.

guysoft, 2019-05-08:
Cool!
Once you get it going you are more than welcome to add it to the list at [here](https://github.com/guysoft/CustomPiOS#list-of-distributions-using-custompios).

WheresWaldo, 2019-05-09:
Still doesn't appear to work under Ubuntu, I am beginning to think it is because Ubuntu (other distros also) obscures or doesn't have a 'root' user. I haven't tried any other distros except for Debian 9, Ubuntu 19.04 and Mint 19.1. Debian is the only one that works 100% of the time, baring stupid commands I put in the chroot_script.

guysoft, 2019-05-12:
@WheresWaldo Can you give the output of
``losetup`` ?

Also note you need the latest dev build.

WheresWaldo, 2019-05-12:
I will git clone the dev build of CustomPiOS, then try my setup. I will try to do it on Monday / Tuesday, since I am busy trying to fix stupid scripting errors in my distribution. I will start with a initial snapshot VM of Ubuntu 19.04 as it appears, once it is broken it is almost impossible to fix.

guysoft, 2019-05-12:
You dont need to clone for the docker-build. just pull from https://hub.docker.com/r/guysoft/custompios/ the devel tag.

Also be sure to run ``losetup``, even if the code does not fix the issue, you should be able to see it. 
If you see files with ``(deleted)`` entry next to them, it means we know that is the issue that is causing the problem.

RyanBalfanz, 2019-05-12:
> You dont need to clone for the docker-build. just pull from https://hub.docker.com/r/guysoft/custompios/ the devel tag.
>
> Also be sure to run `losetup`, even if the code does not fix the issue, you [...]

If this works will you release a new tagged version to Docker Hub? I think when I first tried, I assumed `latest` as I normally would, but then noticed that `1.1.0` existed and it seemed the most appropriate.

RyanBalfanz, 2019-05-12:
I think [this wiki change](https://github.com/guysoft/CustomPiOS/wiki/Building-with-Docker/_compare/289ae8092abbf858fd3989bec220a38f808a339d...4b42f99ec181b9568b7091e22c2b02cf5096b155) might help a bit. There's another `docker-compose.yml` in the repo at the original location which I don't think is the correct file.

guysoft, 2019-05-12:
Yes, if it works it justifies a new version, can be tagged. Would love to
confirm it fixes the issue first.

RyanBalfanz, 2019-05-15:
> `losetup -d /dev/loop7`

```
ubuntu@ip-172-26-7-42:~/CustomPiOS/TestOS/src$ losetup -d /dev/loop7
losetup: /dev/loop7: detach failed: Permission denied
ubuntu@ip-172-26-7-42:~/CustomPiOS/TestOS/src$ sudo losetup -d /dev/loop7
losetup: /dev/loop7: detach failed: No such device or address
ubuntu@ip-172-26-7-42:~/CustomPiOS/TestOS/src$ uname -a
Linux ip-172-26-7-42 4.15.0-1021-aws #21-Ubuntu SMP Tue Aug 28 10:23:07 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

RyanBalfanz, 2019-05-15:
I got this working on my AWS LightSail box (see above) just by using `sudo ./build_dist` (after everything I did above). I will try to do this locally next so that I can actually add some modules and test that I can boot the new image on a Pi.

```
The filesystem is already 328704 (4k) blocks long.  Nothing to do!

+++ losetup -d /dev/loop2
+++ trap - EXIT
+++ popd
~/CustomPiOS/TestOS/src
+++ echo_green -e '\nBUILD SUCCEEDED!\n'
+++ echo -e -n '\e[92m'
+++ echo -e '\nBUILD' 'SUCCEEDED!\n'

BUILD SUCCEEDED!

+++ echo -e -n '\e[0m'
+ exit 0
```

RyanBalfanz, 2019-05-15:
After `apt dist-upgrade` (i.e. not an out-of-the-box fresh 19.04) this is still working as described above.

RyanBalfanz, 2019-05-15:
I cannot get this working on 19.04 (DigitalOcean, fresh 19.04 instance) even with sudo-ing as I mentioned previously on an updated 18.04 box.

```
+++ detach_all_loopback 2019-04-08-raspbian-stretch-lite.img
++++ awk '{ print $1 }'
++++ grep 2019-04-08-raspbian-stretch-lite.img
++++ losetup
++++ losetup -f --show -o 50331648 2019-04-08-raspbian-stretch-lite.img
losetup: 2019-04-08-raspbian-stretch-lite.img: failed to set up loop device: Resource temporarily unavailable
+++ LODEV=
+ exit 1
```

RyanBalfanz, 2019-05-15:
I've not been able to successfully build a simple distro on my macOS host using Docker for Mac and the referenced `Dockerfile` and `docker-compose.yml`. A more complete example with a basic TestOS or something might be helpful.

guysoft, 2019-05-16:
Can you just give an example of plain, with no arguments ``losetup``
In a failed machine?
I see no example of what I asked for in the previous comment.

WheresWaldo, 2019-05-16:
Here is a plain output from 'losetup' on a failed Ubuntu 19.04 build that works 100% on Debian 9.

```
waldo@debian-dev:~/CommunityOS/src$ losetup
NAME       SIZELIMIT OFFSET AUTOCLEAR RO BACK-FILE                                         DIO LOG-SEC
/dev/loop1         0      0         1  1 /var/lib/snapd/snaps/gnome-characters_254.snap      0     512
/dev/loop6         0      0         1  1 /var/lib/snapd/snaps/gnome-logs_61.snap             0     512
/dev/loop4         0      0         1  1 /var/lib/snapd/snaps/gnome-calculator_406.snap      0     512
/dev/loop2         0      0         1  1 /var/lib/snapd/snaps/gnome-3-28-1804_31.snap        0     512
/dev/loop0         0      0         1  1 /var/lib/snapd/snaps/gnome-system-monitor_77.snap   0     512
/dev/loop7         0      0         1  1 /var/lib/snapd/snaps/core18_941.snap                0     512
/dev/loop5         0      0         1  1 /var/lib/snapd/snaps/gtk-common-themes_1198.snap    0     512
/dev/loop3         0      0         1  1 /var/lib/snapd/snaps/core_6673.snap                 0     512

```

RyanBalfanz, 2019-05-17:
I've built (many many times now) my OS on my Mac but have since run into other [problems](https://github.com/guysoft/CustomPiOS/issues/50). I'm using @guysoft's image on docker hub on devel, so I guess this isn't really a problem for me anymore.

guysoft, 2019-05-19:
Ok, found another scenerio where you will get an error "no such file or directory". If you run out of loopack devices!
@WheresWaldo all 7 loopback devices are taken for you. So ti would fail to mount boot.

Try running outside the container:

```
sudo mknod -m640 /dev/loop8 b 7 8
```
Then build again using docker.

Should create ``/dev/loop8``
([taken from here](https://unix.stackexchange.com/a/81257/33742))

guysoft, 2019-05-21:
Ok, pushed a fix, docker pull and try building now.

WheresWaldo, 2019-05-23:
Ok, had a chance to test it today and now it is building properly on Ubuntu. No other changes except using new code from devel.

closed 2019-05-23 by WheresWaldo

guysoft, 2019-05-24:
Great! Thanks for testing and reporting!

Rattkener, 2024-01-28:
Having this issue on a later bookworm PiOS release. No changes made that should have impacted this.  Were there any fixes identified for the host system not being able to use loop devices or is this an entirely unsupported feature now? 

# sudo modprobe loop
# lsmod |grep loop
# losetup -a
# losetup
# uname -a
Linux neo 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

guysoft, 2024-01-28:
@WheresWaldo see #55 that tracks it

Rattkener, 2024-01-29:
Are these not separate issues? #55  seems to be related to issues inside the docker container.

guysoft, 2024-07-18:
@Rattkener sorry for the late reply. I don't know of an issue with that outside a docker container, you might be missing. Open a separate issue and post the exact error messages you are getting.

commit 1519e8ec (2019-05-24): Fix #46