guysoft/CustomPiOS #47: [?] Why does QEMU display error: Unknown host QEMU_IFLA type

WheresWaldo, 2019-04-22:
Running a start_chroot_script that appears to build correctly. In several locations in the script QEMU displays Unknown host QEMU_IFLA type: followed by a number 40, then the same message followed by a 41. It is notable when doing anything within the script that needs to access a file on another site, such as the git repositories or googleapis, etc.

I am using debian 9 in an Oracle VirtualBox

This is not a bug report, but more of a question about why it may be happening, and maybe a suggestion on how to clean it up.

guysoft, 2019-04-23:
I get this too, no idea what it is, seems to work fine.

WheresWaldo, 2019-04-23:
Okay, I am going to ignore it for now and close this issue.

closed 2019-04-24 by WheresWaldo