guysoft/CustomPiOS #50: Filesystem files not available in custom OS image (burned and booted to a Pi)

RyanBalfanz, 2019-05-17:
Here's a (as of yet non-existent) PR showing an example of what I'm doing: https://github.com/guysoft/CustomPiOS/compare/devel...RyanBalfanz:supervisor?expand=1.

The files `/etc/supervisor/conf.d/*` do not exist in my build and deployed custom OS, but I'm not sure why.

guysoft, 2019-05-19:
When are you looking for them, the unpack for that root is at the end:
https://github.com/RyanBalfanz/RTSPiOS/blob/master/src/modules/rtspios/start_chroot_script#L33

Can you provide a full build.log please?

guysoft, 2019-05-19:
I just ran it here and can see it copying to the right place on the build log

```
+ unpack /filesystem/root /
+ from=/filesystem/root
+ to=/
+ owner=
+ '[' 2 -gt 2 ']'
+ cp -v -r --preserve=mode,timestamps /filesystem/root/. /
'/filesystem/root/./README' -> '/./README'
'/filesystem/root/./etc/supervisor/conf.d/inet_http_server.conf' -> '/./etc/supervisor/conf.d/inet_http_server.conf'
'/filesystem/root/./etc/supervisor/conf.d/rtsp-server.conf' -> '/./etc/supervisor/conf.d/rtsp-server.conf'
+ '[' -n '' ']'
+ '[' -d custompios_export ']'
+ rm chroot_script
+ '[' -d filesystem ']'
+ rm -rfv filesystem
```

guysoft, 2019-06-06:
Closing if no reply

closed 2019-07-10 by guysoft