guysoft/CustomPiOS #51: CommunityOS for Robo3D printers

WheresWaldo, 2019-05-29:
[CommunityOS for Robo3D printers](https://github.com/WheresWaldo/CommunityOS)
There are a few very resourceful people on the Robo3D forums that are less than happy that Robo3D has basically abandoned its consumer users. There has only been one single update to their operating environment for the C2 / R2 printers. These printers have a custom UI built using Python and Kivy running on top of OctoPi, with a bunch of junk added. Robo cloned all the repositories for use in their system, which meant users were fully dependent on Robo to provide updates or newer versions of all the OSS parts.

CommunityOS is our attempt to replicate the functionality of Robo's printer OS, but using code that is freely available and easily upgradeable. I based a lot of the code to how OctoPi is set up. There are a lot of additions well beyond what OctoPi does. We have included a large number of OctoPrint Plugins, originally used by Robo in the OS, preconfigured during the CustomPiOS build. We removed the Plugins Robo used that 'phoned home.' Removed CuraEngine but left the code within the module if someone wanted to re-enable its functionality (including automatically adding the new Cura Legacy Plugin). We removed HTTPS access as all modern browsers do not like self signed certs and you cannot register a .local domain with a CA. Touchscreen drivers for various size screens can be selected before building. 

Currently the UI is a customized version of [OctoPrint-TFT](https://github.com/mcuadros/OctoPrint-TFT). The plan is to replace this with a look/work-alike GUI similar to Robo's familiar OS. 

Even though this distribution is geared specifically toward the Robo3D R2 and C2 printers, we have left enough generic choices to make it usable on other 3D printers. I am currently using it on a self built Eustathios as well as a Trony S5. It was designed to not break if you chose to use the interface on other printers,

WheresWaldo, 2019-05-29:
I should add, we are almost at the first phase of deployment. Documentation on how to use the resultin CommunityOS release on the Robo Printer is being written. Once that is done we will post links on the Robo Facebook page as well as their forums.

guysoft, 2019-05-30:
Cool! good to know - however, is this an issue or some sort? Are you missing anything?
Should I close it?

WheresWaldo, 2019-05-30:
Oops, sorry, I meant to comment and close, it was more to let you know that CustomPiOS was being used for another project

closed 2019-05-30 by WheresWaldo