guysoft/CustomPiOS #52: Compile Auto-Hotspot for OctoPi; no success so far?

filipgoc, 2019-05-31:
Hi, 

I'm trying to compile OctoPi with auto-hotspot. I know you're [still working on it](https://github.com/zynthian/zynthian-webconf/issues/34), but the basic functionality seems stable, and I wanted to beta test it. 

I verified that the CustomPiOS has Auto-Hotspot in modules `CustomPiOS/src/modules/auto-hotspot`, and I modified the `OctoPi>src>config` as follows

```
export MODULES="base(raspicam, network, disable-services(octopi,auto-hotspot), password-for-sudo)"
```

It compiles successfully into an image. No errors. OctoPrint runs as expected. 

**Trouble is, I get no hotspot.**

Test 1: I leave wpa-supplicant untouched so that the Pi has no internet, but no hotspots starts. I tried reboot in vain.
Test 2: Also tried connecting the Pi and verify OctoPi is working, then kill the ned, and still nothing. 
Test 3: I'm on RPI 3B+. I tried to add an edimax wifi dongle (that I'm used to used with RPi2 to get wifi) and reboot. This did not help either. 

---

**Q: Did I do all that I needed to do?**
Are there more steps in the compilation or in the setip usage department?

**Q: Should the auto-hotspot work with RPi 3B+ out of the box or do I need a wifi dongle?**

Thanks for any pointers!

guysoft, 2019-06-02:
Hey,
Indeed, there is development in the auto-hotspot in another project using CustomPiOS, called Zynthian.
Their issue is here:
https://github.com/zynthian/zynthian-webconf/issues/34#issuecomment-497279306

It seems that 3 days ago they moved to use a new script and stopped using the auto-hotspot module.
So I guess we could merge that script upstream, I just didn't get to trying it out. If you want to their script it might push this fowards.

filipgoc, 2019-06-02:
Nice. I can give it a try in about one week.

guysoft, 2019-07-31:
@filipgoc 
Any luck?

guysoft, 2019-12-23:
Closing if no reply

closed 2020-01-07 by guysoft

filipgoc, 2020-09-16:
PS: I don't want to re-open this; I just want to apologize for going silent. 
I never made this work.

guysoft, 2020-09-16:
Thanks for the update