guysoft/CustomPiOS #63: Unpack does chmod  on entire target dir

EternityForest, 2019-11-07:
When I unpack directly into a folder that has existing content, that content also gets chmoded, and I believe that process may also mess with the setuid bit.

```bash
cp -v -r --preserve=mode,ownership,timestamps $from/. $to
  if [ -n "$owner" ]
  then
    chown -hR $owner:$owner $to
  fi
```
Seems to be the issue.

I am trying this untested workaround, but I do not like it because of the extra disk activity:
```bash
function unpack() {
  # call like this: unpack /path/to/source /target user -- this will copy
  # all files & folders from source to target, preserving mode and timestamps
  # and chown to user. If user is not provided, no chown will be performed

  from=$1
  to=$2
  owner=
  if [ "$#" -gt 2 ]
  then
    owner=$3
  fi
  mkdir -p /tmp/unpack/
  # $from/. may look funny, but does exactly what we want, copy _contents_
  # from $from to $to, but not $from itself, without the need to glob -- see 
  # http://stackoverflow.com/a/4645159/2028598
  cp -v -r --preserve=mode,timestamps $from/. /tmp/unpack/
  
  if [ -n "$owner" ]
  then
    chown -hR $owner:$owner $to
  fi

  cp -v -r --preserve=mode,ownership,timestamps /tmp/unpack/. $to
  rm -r /tmp/unpack

}
```

guysoft, 2019-11-07:
Yes, seems to be an issue. 
You might be able to save most of the disk activity if you use mv. But I dont know if mv preserves timestamps mode and ownership. It might. Unlike cp mv just changes the innode so it should not be a big addition as cp is.

I would love a PR if you have the time for that.

EternityForest, 2019-11-07:
This is slightly out of my depth (I haven't used Bash as an actual
programming language in a long time) but there seems to be a glob
workaround in a stack overflow post

mv -- * .[!.]* ..?* /dest/

That handles dotfiles semi-correctly without including . and ..

Perhaps three separate commands for normal, dot, and dotdot files?

Dear acts differently with and without trailing slash, I think we want an
error if it doesn't exist and not to clobber seat if it's actually a file,
 so we'd need a way to normalize to one single trailing slash too?

guysoft, 2019-12-23:
Fixed with #65

closed 2019-12-23 by guysoft