guysoft/CustomPiOS #67: Possible leakage of login name from host system?

EternityForest, 2020-01-27:
While building an image, I got the following error:

`+ gpg --fetch-keys https://neilalexander.s3.dualstack.eu-west-2.amazonaws.com/deb/key.txt
gpg: Fatal: can't create directory '/home/daniel/.gnupg': No such file or directory
`
I'm not sure what the cause is, but I suspect that anything that looks for the login name is returning the actual user running the build script outside of the VM. Would setting the login name to  `pi` be more appropriate?

guysoft, 2020-01-27:
What method are you using to build?

EternityForest, 2020-01-27:
I'm just running the build script, with "sudo ./build_dist"

guysoft, 2020-01-27:
I suggest you try the docker build version (vagrant should also work). That should take care of those issues. I can't support building without it at this point.

Also, you didn't answer my second question - what image are you trying to build? Or from which module this error is coming from.

EternityForest, 2020-01-27:
Ah, for some reason Gmail didn't even show a second question. That's rather
odd.

The module is a custom one called netcfg, which is part of
https://github.com/EternityForest/EmberOS

The GitHub code doesn't have the issue, I switched to a workaround pretty
quickly and never committed the failing code, and it's probably not worth
it to dig through the whole distro for it.

Still, it might be relevant to someone someday, so documenting it in an
issue seemed like a good idea.

guysoft, 2020-01-28:
> Ah, for some reason Gmail didn't even show a second question. That's rather
odd.

It might be that I updated it on github comment and it sent your the email before it was updated.

Ok now I can see what you are building, neat!
You imported CustomPiOS in to your project in a non-standard way, that might be some of the causes. I really suggest you work with something to encapsulate the environment. Both Vagrant and Docker are supported, with rising popularity of Docker since its faster. I would recommend you try and use that. Your system might have non-standard settings which I can't debug from a distance. Docker/Vagrant come and fix that.

Also, you should not really maintain CustomPiOS at the side of your project. If you really need a fork I should recommend forking CustomPiOS, it would let you follow changes in this repo and see if there are changes worth merging in. Comparing what you have and what has been added would be a pain with your current setup.
Also, you are writing a pretty big list of modules, if you fork you could propose them as pull requests and they would get merged to mainline, meaning people could use and improve them for you.

guysoft, 2020-03-03:
Closing if no reply

closed 2020-03-05 by guysoft