guysoft/CustomPiOS #71: Build fails because of line added in "start_chroot_script"

AlbertMN, 2020-04-22:
Hi there. I am _very_ new to this, so my apologies if this is an ignorant question, but I couldn't Google my way to a solution 😅

I'm trying to build a quite simply image. I'm building it from Octopi _(having forked it)_. Were originally gonna create a `variant` of OctoPi, as per suggestion by the developer, but  as this image needs to be done very soon, I just went with modifying OctoPi instead.

All I have modified is the `src/modules/octopi/filesystem/home` folder, where I have added a folder containing some Python scripts. These are added to the image just fine, but when adding the line `sudo -u pi pip install pyyaml requests python-crontab` _(dependencies of my scripts)_ in the `start_chroot_script`, the build fails consistantly.

Is there a better way of installing these dependencies in the CustomPiOs build? Just need the required pip packages installed, and for one of my python-scripts to be opened at first boot of the RPI.

I think all I need is being pointed in the right direction (hopefully). Thanks!

guysoft, 2020-04-22:
Hey, 
1. A variant should not take longer to build
2. Please attach your ``build.LOG``  so I can see what your actual error message is.

AlbertMN, 2020-05-02:
Sorry it's taken so long for me to get back to you. Looking at this again in the next couple of days! :)

closed 2020-05-24 by AlbertMN