guysoft/CustomPiOS #72: Building Broken on x86\64 Debian System

asdf1nit, 2020-04-23:
Commented on the issue here 
https://github.com/guysoft/CustomPiOS/commit/fb4b81020f4a7fb45be15bc0ff75f02a1885e5b3#r38695924

Using a custompios file prior to this commit or just editing that line has building working again.

Will provide a fix as soon as I have one that doesn't break any work after this commit.

guysoft, 2020-04-23:
Interesting catch.
It might be fallout to adding support building from ARM, will try find time to investigate.

Basically the logic is - if its a amd64 system then you need qemu there. If its some version of arm then you don't.

asdf1nit, 2020-04-23:
The output of the arch command on my system is
```
jonathan@buster:~/Documents/Guysoft/FullPageOS/src$ arch
x86_64
```

guysoft, 2020-04-23:
Ok, I think I got it is ``$BASE_ARCH`` unset or is it set to something?
It seems like there is a missing combination there. The code today to figure if qemu should be copied is:

```
  if [ -f /.dockerenv ] && [ "$(arch)" != "armv7l" ] && [ "$(arch)" != "aarch64" ] ; then
    if [ "$BASE_ARCH" == "armv7l" ]; then
      cp `which qemu-arm-static` usr/bin/qemu-arm-static
    elif [ "$BASE_ARCH" == "aarch64" ] || [ "$BASE_ARCH" == "arm64" ]; then
      cp `which qemu-aarch64-static` usr/bin/qemu-aarch64-static
    fi
  fi
```

guysoft, 2020-04-23:
@asdf1nit are you building useing the docker build method? Because otherwise the issue might be in this if:
```
if [ -f /.dockerenv ] && [ "$(arch)" != "armv7l" ] && [ "$(arch)" != "aarch64" ] && [ "$(arch)" != "arm64" ] ; then

asdf1nit, 2020-04-24:
I'm not sure if `$BASE_ARCH` is set I've been looking through the code to find where it's supposed to get set at and haven't found it yet.

I've found the code that outputs the error code you mentioned above and that's what I've changed to get it to build again. Those 2 specific block's seem to be where the problem lies. Once I find where `$BASE_ARCH` is and confirm it is set, I'll see what I can do in those blocks. Seems that the second block the outer if statement 
```if [ -f /.dockerenv ] && [ "$(arch)" != "armv7l" ] && [ "$(arch)" != "aarch64" ] && [ "$(arch)" != "arm64" ] ; then```
is where the issue lies. If I'm correct since I'm not using docker then this fails and I just skip to the elif part which isn't the branch I want to go down. I'll do some testing tonight and see what I come up with. If you can point me to where `$BASE_ARCH` is set I'll go over that and make sure it's not empty\null.
 
I'm not using the docker method. Just haven't gotten around to figuring  out how I would need to build it yet.

guysoft, 2020-04-29:
That block says "if docker". So I guess we could try remove that.
I fixed that, please try building from master and tell me if this works.

closed 2020-04-29 by guysoft

asdf1nit, 2020-05-05:
So this is working now but I'm still a bit confused. This is in my build log file
```
'/home/jonathan/Documents/Guysoft/CustomPiOS/src/modules/gui/filesystem/root_init/usr/share/xsessions/guisession.desktop' -> './filesystem/root_init/usr/share/xsessions/guisession.desktop'
++ arch
+ '[' x86_64 '!=' armv7l ']'
++ arch
+ '[' x86_64 '!=' aarch64 ']'
+ '[' armv7l == armv7l ']'
++ which qemu-arm-static
+ cp /usr/bin/qemu-arm-static usr/bin/qemu-arm-static
+ cp /home/jonathan/Documents/Guysoft/CustomPiOS/src/modules/gui/start_chroot_script chroot_script
+ chmod 755 chroot_script
+ cp /home/jonathan/Documents/Guysoft/CustomPiOS/src/common.sh common.sh
+ chmod 755 common.sh
+ '[' -f /.dockerenv ']'
+ echo 'Building on ARM device a armv7l/aarch64/arm64 system, not using qemu'
Building on ARM device a armv7l/aarch64/arm64 system, not using qemu
+ chroot . /bin/bash /chroot_script
+ set -e
+ source /common.sh
+ unpack /filesystem/home/pi /home/pi pi
+ from=/filesystem/home/pi
+ to=/home/pi
```

specifically "Building on ARM device a armv7l/aarch64/arm64 system, not using qemu" which is not the case. Is this just commented wrong? Shouldn't this be using qemu since I'm not on an arm device?

guysoft, 2020-05-05:
Ok, I know why. Pushed a fix, try now.

commit 8da4e4eb (2020-04-29): Fixes #72