guysoft/CustomPiOS #73: pi 4 boot config.txt

asdf1nit, 2020-05-03:
With the update to the pi 4 board there is now another

guysoft, 2020-05-03:
Thanks,  keep me posted!  I have no hardware to test that!

asdf1nit, 2020-05-03:
Ok just got finished testing. Boot will boot with out any modification to the config.txt so the following block can be removed.
```
    check="$(sudo cat /proc/cpuinfo | grep 'Revision' | awk '{print $3}' | sed -e 's/[a-c]//')"
    if [ "${check}" != "03111" ]; then
        sudo sed -i /boot/config.txt -e "s/^\#dtoverlay=vc4-kms-v3d/dtoverlay=vc4-kms-v3d/"
        printf "dtoverlay=vc4-kms-v3d\n" | sudo tee -a /boot/config.txt
    fi
```
The build was with gpu acceleration enabled. I can rebuild without gpu and test it later this week if you want to hold off but I'm not sure it's going to matter. Looks like the fksm driver is the preferred choice now and included in the default debian lite image on all variants of the pi based on the research I did to see if there was any newer preferred drivers.

guysoft, 2020-05-03:
That code takes in to account older pis too. Might e better to add > than . Assuming all newer pis wont need it.

closed 2020-05-06 by guysoft