guysoft/CustomPiOS #76: Execute functions in variant config file & "pip: command not found"

AlbertMN, 2020-05-24:
Hi there :)

I have two questions/issues, but first some info on the project;
I'm building a variant for OctoPi (forked the OctoPi repository), so I am a little unsure whether I should post this here or at the OctoPi repository, but my guess is here, since it's more of a CustomPiOs question.

The variant is quite simple, some files are included in the build _(which I've got working)_, I wanna edit the hostname and want some files to be executed upon first boot.

________

**Questions;**

**1.** In my variant I have a `config` file taken straight out of the CustomPiOs variant example, but I attempt to set the hostname to _"static_text-{4 random characters}"_ like this;
`BASE_OVERRIDE_HOSTNAME="SimplyPrint-$(cat /dev/urandom | tr -dc 'a-zA-Z0-9' | fold -w 4 | head -n 1)"`
The hostname still ends up being "octopi" though, and when just having the value be; `BASE_OVERRIDE_HOSTNAME=blueberrypi` it worked fine. Is it not possible to execute functions / run code in the hostname?

**2.** In `post_chroot_script` I tried doing the following;
```
# Install required python packages
yes | pip install pyyaml requests python-crontab

# Make sure the SimplyPrint system runs
python /home/pi/SimplyPrint/crontab_manager.py -s > /dev/null 2>&1

# Install SimplyPrint OctoPrint plugin
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install /home/pi/SimplyPrint/SimplyPrint-OctoPrint-Plugin.zip --no-cache-dir && \\
rm -rf /home/pi/SimplyPrint/SimplyPrint-OctoPrint-Plugin.zip
```

right after unpacking the variant filesystem, but if any of these lines exist, the build fails _([build.log](https://github.com/guysoft/CustomPiOS/files/4673697/build.log))_. Is this the right place to do what I'm attempting to do? I just want it to install the dependencies and possibly run a script once. This script could also just run on first boot of the Pi. I am also trying to install an OctoPrint plugin on boot from a zip file included in the filesystem.

The last lines of the terminal when the build fails is;
```
'/tmp/unpack/./home/pi/scripts/welcome' -> '/./home/pi/scripts/welcome'
+ rm -r /tmp/unpack
+ yes
+ pip install pyyaml requests python-crontab
/chroot_script: line 11: pip: command not found
++ echo_red -e '\nBUILD FAILED!\n'
++ echo -e -n '\e[91m'
++ echo -e '\nBUILD' 'FAILED!\n'

BUILD FAILED!

++ echo -e -n '\e[0m'
```
with line 11 being the "`yes | pip install (...)`" line.

Seeing the build fails I'm thinking the `post_chroot_script` might not be the right place to do what I'm attempting to do, so any advice on how to proceed from here would be greatly appreciated! :)

guysoft, 2020-05-24:
1. It looks like the hostname is set with the random chars you wanted. Taken form the build log:
```
++++ BASE_OVERRIDE_HOSTNAME=SimplyPrint-HXCg
```
And then again you can even see it running the sed command that replaces the name in the host name:
```
+ sed -i -e s@raspberrypi@SimplyPrint-HXCg@g /etc/hosts
```
Note this only happens once in the build, when the image is build, you are not going to get a new hostname each time you flash the image, you will need a script on the device to do that.

2. The error fails because pip does not exist. 

The only pip installed is python-pip-whl which I don't know how it behaves outside a virtual environ.
Suggest you do something like this, as done in the docker module:
```
    wget https://bootstrap.pypa.io/get-pip.py -O - | python3
    pip3 install --ignore-installed PyYAML
    pip3 install requests
    pip3 install python-crontab

[source of the lines](https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/docker/start_chroot_script#L33)

You could use python2, but if you are writing something new, better use python3 since python2 is at End Of Life, OctoPi is also going to move to python3 in the next release.

AlbertMN, 2020-05-24:
Ah right... It sets the hostname once - will just have it execute on first startup then _(which you do exactly how btw?)_

Regarding **2.** I'll have a look and get back to you! Thanks :D

guysoft, 2020-05-24:
1. I'd use ``hostnamectl`` which is a command that lets you do stuff like that. And put it in an systemd service. You can see a systemd example in CustomPiOS ``usage-statistics`` module. All it does is add a command that executes on startup using systemd: https://github.com/guysoft/CustomPiOS/tree/devel/src/modules/usage-statistics

AlbertMN, 2020-05-31:
Have now migrated the entire project to Python 3, and the pip3-install code you sent above worked like a charm! All the dependencies are now installed by default, but can't seem to get the `start_chroot_script` to run?

I'm building a variant, so the `start_chroot_script`, I try to do the following;
```
search=octopi && \\
if grep -q \$search \"/etc/hostname\"; then
  replace=\"SimplyPrint-\$(cat /dev/urandom | tr -dc 'a-zA-Z0-9' | fold -w 4 | head -n 1)\" && \\
  sed -i \"s/\${search}/\${replace}/g\" /etc/hostname && \\
  sed -i \"s/\${search}/\${replace}/g\" /etc/hosts && \\
  echo \"SimplyPrint installed - rebooting to change hostname\"
  sudo reboot
fi

sudo python3 /home/pi/SimplyPrint/crontab_manager.py -s > /dev/null 2>&1
```

This should run on startup, and just make sure that if the hostname is `octopi`, it's replaced by a _(more likely to be-)_ unique hostname, as well as running a script that makes sure the cron jobs are set up.

Did I mistake `start_chroot_script` for something it's not? I simply want the code above ^ to run when the Raspberry Pi starts _(after being flashed - if it only runs on first statup, that'd be the best case, but running it on every startup isn't a big issue)_.

Thanks so much for your help so far! Love this project :)

guysoft, 2020-05-31:
Hey, yes you did misstake start_chroot_script.
``start_chroot_script`` is a script that runs when you build the image. The process in CustomPiOS is:
1. it takes the base image opens it
2. Edits it
3. closes it.

The way it edits it is using modules. And you can also set a variant which is a bit like an extra module on top. It lets you change the image and have your own variant.
A variant structure is a little like a module, which has a ``start_chroot_script``, and ``end_chroot_script`` and a ``filesystem`` folders, [see what a module is here](https://github.com/guysoft/CustomPiOS/wiki/Modules)

So this stuff is running when you are creating the image, on the machine that you are building the image with, it does not run on the raspberrypi/device that you flash the image to.
To do that you need to use linux/debian services. And the service on a Debian system that controls startup scripts is systemd. I gave you an example module that adds something to systemd that runs on boot in https://github.com/guysoft/CustomPiOS/issues/76#issuecomment-633293108

AlbertMN, 2020-06-01:
Makes a lot more sense now - simply missed that you didn't refer to the `start_chroot_script` in your pervious comment, but the service-stuff.

I have written the service and gotten it to work locally, but I can't seem to get it to start in the image? I've tried the following file-structure;

**Like the `usage-statistics` module**
- filesystem
  - root
    - etc/systemd/system
      - simplypi.service
    - usr/bin
      - simplyprint_start.sh

This didn't work - the service could not be found. I then looked at how the OctoPi module did it;

**Like OctoPi**
- filesystem
  - home
    - root/bin
      - simplyprint_start.sh
  - root
    - etc/systemd/system
      - simplypi.service

This didn't work either - when typing `sudo systemctl enable simplypi.service` it says that it does not exist - like the previous try. Previous to this try, the filetree looked like this;

- /etc/systemd/system
  - simplypi.service
- /usr/bin/simplyprint_start.sh

With this the service wasn't enabled by default by the image, but I was able to enable it manually from CLI. It worked as it should _(well, almost, it didn't set the hostname or restart the system - for this I had to execute the file manually, but betting this isn't a CustomPiOs-related thing)_.

**`simplypi.service`**
```
[Unit]
Description=SimplyPrint startup service

[Service]
Type=oneshot
ExecStart=bash /usr/bin/simplyprint_start.sh

[Install]
WantedBy=multi-user.target
```

**`simplyprint_start.sh`**
```
#!/bin/sh

python3 /home/pi/SimplyPrint/crontab_manager.py -s > /dev/null 2>&1

search=octopi
if grep -q $search "/etc/hostname"; then
  replace="SimplyPrint-$(cat /dev/urandom | tr -dc 'a-zA-Z0-9' | fold -w 4 | head -n 1)"
  sed -i "s/${search}/${replace}/g" /etc/hostname
  sed -i "s/${search}/${replace}/g" /etc/hosts
  reboot
fi
```

*`start_chroot_script`*
```
systemctl enable simplypi.service
```

I'm not quite sure what is wrong with the current setup. Thanks again for your help!

guysoft, 2020-06-01:
Check:
0. providing ``build.log`` would help me see what is going wrong, without it  I am only guessing.
1. if the files are actually reaching the system. You might need to add the ``unpack`` command. Like here:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/usage-statistics/start_chroot_script#L15

2. Make sure you are run the enable command.
Like here:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/usage-statistics/start_chroot_script#L21

3. There might be an issue, I am not 100% sure. That the build conflicts somehow with the service, because its also trying to bring it up (depends on the service settings.
After you made sure 1 and 2 steps are ok, you can try adding 
```
apt-get update
if [ $( is_in_apt policykit-1 ) -eq 1 ]; then
  sudo apt-get -y install policykit-1
fi

# prevent any installed services from automatically starting
echo exit 101 > /usr/sbin/policy-rc.d
chmod +x /usr/sbin/policy-rc.d

[[[ insert your systemd enable comand here ]]

rm -r /usr/sbin/policy-rc.d || true
```
This disables systemd from starting. Its taken from the ``disable-services`` module here:
https://github.com/guysoft/CustomPiOS/tree/devel/src/modules/disable-services

When you write modules you can add braces that around cases like that, but there is a limitation in variants that you can wrap modules around them.
(What I mean by wrapping is that in ``config`` file you have ``MODULES=disable_services(your_module)`` 
Thats why there are ``start_chroot_script`` and ``end_chroot_script``. They run before and after.
workaround now is to add those 5 lines manually.

AlbertMN, 2020-06-03:
I don't think there is a confict, and all the files are actually reaching the system. As it is right now, the files are as follows;

- variants/SimplyPi
  - root/etc/systemd/system/simplypi.service
  - home/root/bin/simplyprint_start.sh

Both files files are in the build, but upon doing `systemctl enable simplypi.service` I still get _"(...) service does not exist"_, and it doesn't get enabled by the system. Are the files actually in the right places?

I only have the `unpack` command in the `post_chroot_script`, but I figured that was enough.

Here's the [build.log](https://github.com/guysoft/CustomPiOS/files/4721772/build.log).
This is the variant I'm trying to build; [SimplyPi.zip](https://github.com/guysoft/CustomPiOS/files/4721781/SimplyPi.zip)

AlbertMN, 2020-06-11:
@guysoft that makes sense - thank you! Since your comment I have built the image 4 times, and in this last one it's very close to working.

The files are in the correct locations, the "simplypi.service" shows as "enabled" and when checking its status after first boot of image, it says it was loaded and that it executed;

```
● simplypi.service - SimplyPrint startup service
   Loaded: loaded (/etc/systemd/system/simplypi.service; enabled; vendor preset: enabled)
   Active: activating (start) since Thu 2020-02-13 16:10:07 GMT; 3 months 27 days ago
 Main PID: 292 (sudo)
    Tasks: 6 (limit: 2077)
   Memory: 8.8M
   CGroup: /system.slice/simplypi.service
           ├─292 /usr/bin/sudo bash /home/root/bin/simplyprint_start.sh
           ├─317 bash /home/root/bin/simplyprint_start.sh
           ├─933 bash /home/root/bin/simplyprint_start.sh
           ├─934 cat /dev/urandom
           ├─935 tr -dc a-zA-Z0-9
           └─936 fold -w 4

Feb 13 16:10:07 octopi systemd[1]: Starting SimplyPrint startup service...
Feb 13 16:10:07 octopi sudo[292]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/bash /home/root/bin/simplyprint_start.sh
Feb 13 16:10:07 octopi sudo[292]: pam_unix(sudo:session): session opened for user root by (uid=0)
```

but the stuff that the `simplyprint_start.sh` script should do was never done. The start-script is quite basic and only has 11 lines;

```
#!/bin/sh

python3 /home/pi/SimplyPrint/crontab_manager.py -s > /dev/null 2>&1

search=octopi
if grep -q $search "/etc/hostname"; then
  replace="SimplyPrint-$(cat /dev/urandom | tr -dc 'a-zA-Z0-9' | fold -w 4 | head -n 1)"
  sed -i "s/${search}/${replace}/g" /etc/hostname
  sed -i "s/${search}/${replace}/g" /etc/hosts
  reboot
fi
```

The `crontab_manager.py` script was never executed, and the hostname never changed. If the Python script had been executed, there would have been two cron jobs in the `crontab`, which remains untouched. When _just_ doing; `bash /home/root/bin/simplyprint_start.sh` which seems to be the line that the service executes, the output is;
```
sed: couldn't open temporary file /etc/sed3LjSjf: Permission denied
sed: couldn't open temporary file /etc/sedcWWe0f: Permission denied
Failed to set wall message, ignoring: Interactive authentication required.
Failed to reboot system via logind: Interactive authentication required.
Failed to open initctl fifo: Permission denied
Failed to talk to init daemon
```

But when doing writing **sudo** in front of the script, it is executed perfectly and the RPi starts up with a new hostname and the cronjobs created.

I thought I could make it work by having the `ExecStart` be `sudo bash /home/root/bin/simplyprint_start.sh`, but this hasn't helped. The whole `simplypi.service` file looks like this;
```
[Unit]
Description=SimplyPrint startup service

[Service]
Type=oneshot
ExecStart=sudo bash /home/root/bin/simplyprint_start.sh

[Install]
WantedBy=multi-user.target
```

Any help on getting the scripts to actually run would be awesome!

Don't think the build log has much to tell here, but just in case; [build.log](https://github.com/guysoft/CustomPiOS/files/4763802/build.log)

guysoft, 2020-06-11:
1. I would write this as:
```
[Unit]
Description=SimplyPrint startup service

[Service]
Type=oneshot
User=root
ExecStart=/home/root/bin/simplyprint_start.sh

[Install]
WantedBy=multi-user.target
```
And make sure you also run after the unpack command, so the script is executable:
```
chmod 755 /home/root/bin/simplyprint_start.sh
```

You don't need sudo because systemd executes as root by default. I am not sure how sudo behaves in that environment.
You can set ``User=root`` in the ``[service]`` section explicitly.

Also I warmly suggest in ``simplyprint_start.sh`` to start with somthing like:

```bash
#!/bin/sh

echo "starting up simplyprint_start.sh"
python3 /home/pi/SimplyPrint/crontab_manager.py -s > /dev/null 2>&1
```
Then when you run ``systemct status simplypi.service`` you should see that line and know that the file does indeed run. Then you can start debugging the rest of your code known that the ``systemd`` side is indeed working, and not search for crontabs and perhaps miss some python error that behaves differently. For debug I would also remove ``2>&1`` because you want to see all the errors you can possibly get when running ``systemctl status simplypi.service``

AlbertMN, 2020-06-11:
Thanks for the swift response! A build with the proposed changes is running now, will report back with the results :)

AlbertMN, 2020-06-11:
It executes the script! Now the issue seems to be that the OctoPrint `/home/pi/.octoprint/config.yaml` file was not yet generated / had any content, so the script failed, but at least it ran! When rebooting the Pi it successfully creates the cronjobs on startup. Now, a workaround for this could just be a Python-code-fix making the script not fail if the API key from the config file cannot be found, but would there be a way to make our service start _after_ the service that creates / fills out the OctoPrint config file?

The hostname is never set though - not on first reboot, not on second or third even though the crontab keeps getting updated perfectly. Doing `sudo /home/root/bin/simplyprint_start.sh` sets the hostname just fine. It says nothing about this in the status of the service, so I'm at a loss of what to do about this one :/

Status of the service just after first boot;
```
pi@octopi:~ $ systemctl status simplypi.service
● simplypi.service - SimplyPrint startup service
   Loaded: loaded (/etc/systemd/system/simplypi.service; enabled; vendor preset: enabled)
   Active: activating (start) since Thu 2020-02-13 16:10:07 GMT; 3 months 27 days ago
 Main PID: 261 (simplyprint_sta)
    Tasks: 5 (limit: 2077)
   Memory: 8.5M
   CGroup: /system.slice/simplypi.service
           ├─261 /bin/sh /home/root/bin/simplyprint_start.sh
           ├─930 /bin/sh /home/root/bin/simplyprint_start.sh
           ├─931 cat /dev/urandom
           ├─932 tr -dc a-zA-Z0-9
           └─933 fold -w 4

Feb 13 16:10:37 octopi simplyprint_start.sh[261]: Traceback (most recent call last):
Feb 13 16:10:37 octopi simplyprint_start.sh[261]:   File "/home/pi/SimplyPrint/crontab_manager.py", line 2, in <module>
Feb 13 16:10:37 octopi simplyprint_start.sh[261]:     __import__('__init__')
Feb 13 16:10:37 octopi simplyprint_start.sh[261]:   File "/home/pi/SimplyPrint/__init__.py", line 762, in <module>
Feb 13 16:10:37 octopi simplyprint_start.sh[261]:     set_config()
Feb 13 16:10:37 octopi simplyprint_start.sh[261]:   File "/home/pi/SimplyPrint/__init__.py", line 719, in set_config
Feb 13 16:10:37 octopi simplyprint_start.sh[261]:     get_octoprint_api_key()
Feb 13 16:10:37 octopi simplyprint_start.sh[261]:   File "/home/pi/SimplyPrint/__init__.py", line 201, in get_octoprint_api_key
Feb 13 16:10:37 octopi simplyprint_start.sh[261]:     the_key = the_config["api"]["key"]
Feb 13 16:10:37 octopi simplyprint_start.sh[261]: KeyError: 'api'
```

guysoft, 2020-06-12:
You could set that your service starts after OctoPrint. I would ask @foosel if she has any tips.
This is sort of going beyond the issue cope here.

I would recommended trying:
1. Generate your own yaml before anything starts.
2. You can run you script after OctoPrint starts by telling systemd yo start it after Octoprint, here is an example for mysql: 
https://github.com/guysoft/AlarmBot/blob/master/src/config/alarmbot.service#L3
3. You could make the script wait with a while loop and checks if os.path.file until the file is there.

AlbertMN, 2020-06-12:
You're absolutely right - the original issue has been resolved, so I'll close this issue and play around with solutions and get some more systemd-specific advice somewhere else!

Thank you so much for your help - has brought us a lot closer to a fully working image :)

closed 2020-06-12 by AlbertMN

guysoft, 2020-06-15:
BTW, just saw your fork on OctoPi.
Warmly suggest you add meanifgul commit messages other than just "fix", "fix2".
That way you can understand what you did along the way, and also others can learn what you are doing too.

AlbertMN, 2020-06-15:
Yeah the commit messages aren't the best right now ;P Since it's just one line or a few characters modified at a time right now, there's not much meaning to be given to each commit (and them being such a short time apart, trying to fix the same thing). Meaningful commit messages will be a thing when the project is in the next phase :)

guysoft, 2020-06-15:
Ok, if that is the case I suggest you make all those commits to a single branch, and then squish-merge them with a meaningful commit over all those small increments.
From experience, if you just commit "fix", it will eventually work, and you will not want to ever touch it and it will remain that way.
You can't see in a commit log that its indeed one line - you can only see the text of your commit message.
Thats IMHO, you are welcome to do as you see fit, it will make it hard for me and others to see what you actually did.

AlbertMN, 2020-06-15:
Thanks! Duly noted 😊

AlbertMN, 2020-06-15:
Just a little update on the hostname-issue, if anyone has the same issue; the issue was the line; `replace="SimplyPrint-$(cat /dev/urandom | tr -dc 'a-zA-Z0-9' | fold -w 4 | head -n 1)"`. It had a hard time doing the `cat` command, which might be a result of some path-issues. Replaced it with `openssl rand -hex 2` instead, since `openssl` is included in the image anyways. The hostname now changes dynamically just fine!