guysoft/CustomPiOS #77: ArchLinux build issue: unmount detaches loopback

MaStr, 2020-05-24:
Hi,
thank you for your framework. Very interesting and charming work.

I am currently investigating the possibilities and try a "hello world" build. My host system is running ArchLinux and it seems the upstream tool behavior is different then expected: During unmount of a loopback mount, my loopback device is detached, too. As an result, the framework fails because it is not able to detach its own lo-devices.
(I stumbled on similar issues on a different project).

Where does it make sense to add a build option to ignore the detach in some cases? 
The relevant function - to be enhanced- seem to be: "common.sh" 

best regards Matthias

guysoft, 2020-05-24:
Hey @MaStr ,
It might be a good idea to use docker, or even vagrant to build if you are using ArchLinux. The build env was only tested on Debian/Ubuntu. Luckily the docker env was built just for this situation, where your distro isn't the one I am using. In fact, I am using it now even when I build inside Ubuntu/Debian.
See https://github.com/guysoft/CustomPiOS/wiki/Building-with-Docker

MaStr, 2020-05-27:
Building with docker works (I added some remarks on the wiki page).
(Still want to have a native version ;-) )

closed 2020-05-27 by MaStr

guysoft, 2020-05-27:
If you want a naitive version you are welcome to support it.
But I recommended docket because it handles better the loopback mounts. If something goes bad you can delete the container.