guysoft/CustomPiOS #78: github action for docker build

jeffeb3, 2020-06-01:
I am still using custompios, and the octopi work to make the [v1pi](https://jeffeb3.github.io/v1pi/) images. It is working well. I basically ignore it most of the time, and then about every three months I come back and spin up a new pair of images from the scripts on my local machine. 

My setup is still a bit brittle though. I am using my own docker setup, and it works on this machine, but it would be better if it was done nightly someplace.

I was thinking about working on some github actions, or a circleci, or travisci setup. But I don't want to reinvent the wheel. Do you (or any images you know) have a CI setup like that? I've used the github actions for another project, and circle CI for another. They both seem to work pretty well. I'm not sure if it is possible with the loop devices though.

jeffeb3, 2020-06-11:
OK. I have been distracted since I asked. I haven't done anything on it. That link is useful. The script is all in github, thankfully. You can see the steps it takes here:

https://github.com/bitbyt3r/FullPageOS/blob/devel/.github/workflows/main.yml#L13

The first step is running the .sh script, which is here:
https://github.com/bitbyt3r/FullPageOS/blob/devel/.github/workflows/main.yml#L13

So it isn't setting up docker to do the work. Github is running it somewhere, and they must be ok with the script calling `sudo modprobe loop`. You can see the results here:
https://github.com/bitbyt3r/FullPageOS/actions

It looks like it's working, and it is working in some kind of virtual environment. I'm not sure if that means container or more full fledged environment, but if it's working for that project, it should probably work for any of ours.

guysoft, 2020-06-12:
Coolx please document what you do. We could add it to the wiki
 Perhaps even I could move some builds there.

guysoft, 2020-06-12:
@bibytt3r also. If you can give any insight or documentation that would be great.

bitbyt3r, 2020-06-12:
Sorry I lost track of your earlier request! It fell to the bottom of my TODO list when things got busy at work. I created a pull request to FullPageOS: https://github.com/guysoft/FullPageOS/pull/332

That should also be fairly generic, so it should be able to build similar projects based on CustomPiOS without changes.

I also created a draft pull request to add that action to v1pi, (https://github.com/jeffeb3/v1pi/pull/24) however it isn't working yet. I haven't figured out how to set OCTOPI_INCLUDE_HAPROXY correctly.

bitbyt3r, 2020-06-12:
You can see the successful build for FullPageOS here: https://github.com/bitbyt3r/FullPageOS/actions/runs/133097110

And the most recent failed build for v1pi here: https://github.com/bitbyt3r/v1pi/actions/runs/133106807

I'll keep poking the build for v1pi for a bit and let you know if I figure out what's missing.

guysoft, 2020-06-28:
@jeffeb3 Any news?
There is a PR but I am not sure I like the fact that only after you tag a new version you get a build, since you want to release after you built and tested an image.

bitbyt3r, 2020-06-28:
I can change the rules for which commits trigger a build if you like. Would you like every push to trigger a new build? It wouldn't show up on the releases page, instead you'll have to download the build from the artifacts drop down on  the action page.

guysoft, 2020-06-28:
@bitbyt3r Yes that would be better. It would be scrabble and we could in the future post it to something like: https://gnethomelinux.com/OctoPi/nightly/ (broken will fix it).
The code that generated it is here: https://github.com/guysoft/OctoPi/blob/devel/src/nightly_build_scripts/generate_nightly_page.js

bitbyt3r, 2020-06-28:
Ok, I updated the pull request on FullPageOS to run a build on every push, and disabled creating releases automatically.

guysoft, 2020-06-28:
Cool - it might be good to have a periodic nightly build too. @foosel mentioned that is possible.
Current version: https://github.com/guysoft/FullPageOS/blob/devel/.github/workflows/main.yml

jeffeb3, 2020-06-28:
Glad you're making progress. I have been... Distracted.

guysoft, 2021-02-11:
It seems like there is a working one on the OctoPi github:
https://github.com/guysoft/OctoPi/actions

I am not using it for nighties because I need them uploaded over FTP and such to mirrors. But it seems this issue is solved, but documentation should be added.

I think the only thing missing here is to extract the distro name that is hardcoded to "octopi":
https://github.com/guysoft/OctoPi/blob/devel/.github/workflows/build.yml

jeffeb3, 2021-02-11:
Just pointing to an example of two is a good first step in documentation. The users are going to be advanced users anyway. 

I am happy with my workflow, although I am going to copy that apt fix. From my perspective, this is fine to close anytime.

guysoft, 2021-02-14:
Done, thanks for all the help!

commit cd75b84c (2021-02-14): Add example of CI/CD github actions, fixes #78
closed 2021-02-14 by guysoft