guysoft/CustomPiOS #82: Variant build fails with 'Error: could not find image: '

raymondh2, 2020-09-25:
Building in Ubuntu 18.04 with QEMU setup

https://github.com/raymondh2/MainsailOS/
[build.log](https://github.com/guysoft/CustomPiOS/files/5285508/build.log)

guysoft, 2020-09-26:
The log says its looking for an armbian image at:
``/home/adminlocal/MainSailOS/src/workspace-armbian``
what files are in that folder?

raymondh2, 2020-09-26:
Maybe I'm misunderstanding how the build process works for variants. I assumed you need an image in the image/ directory. Does this need to be placed elsewhere?

raymondh2, 2020-09-26:
Right I tried placing the Armbian image in the workspace-armbian directory and got this instead. 
[build.log](https://github.com/guysoft/CustomPiOS/files/5287188/build.log)

guysoft, 2020-09-30:
Hey, you didn't answer my question:  what files do you have in that folder?
because I can see its searching for "Armbian_20.08.1_Orangepipcplus_buster_current_5.8.5.img" in the log.
The issue is that its not finding the image, likely because its a different name, or its not their at all. But without an answer to the question its not possible for me to know

raymondh2, 2020-09-30:
That is the file in the directory. I had placed it in there prior to starting the build process.

guysoft, 2020-09-30:
The image should be either zip or 7z as shown in this line:
https://github.com/guysoft/CustomPiOS/blob/devel/src/variants/armbian/config#L2
```
BASE_ZIP_IMG=`ls -t ${DIST_PATH}/image-variants/*.{zip,7z} | head -n 1`
```
You can either overload this variable with config.local file or fix yoru image name so it gets detected.

raymondh2, 2020-09-30:
So when using the Armbian variant the BASE_ZIP_IMG variable isn't being populated. Even when placing BASE_ZIP_IMG=`ls -t ${DIST_PATH}/image-variants/*.{zip,7z} | head -n 1` in the config file. 
[build.log](https://github.com/guysoft/CustomPiOS/files/5306416/build.log)

guysoft, 2020-10-01:
You need to place a zip or 7z file, look at the line and see its looking for an archive to extract.
I didn't say "go paste that line". I wrote "The image should be either zip or 7z as shown in this line".

closed 2020-10-02 by raymondh2

guysoft, 2020-10-02:
Hey, you just closed the issue with no feedback.
Have you solved the issue? And if so please share so others finding this issue will know what to do.

raymondh2, 2020-10-02:
Apologies. I swear I hit close and comment. The issue was me placing the image of Armbian in the wrong image folder. Image vs image-variant.