guysoft/CustomPiOS #86: Make host directory visible from target

Salamandar, 2020-11-24:
Would it be possible to bind mount a directory on the host to make it visible on the target ? 
Right now I have a wrapper around `build_dist` that copies a bunch of files from my git repo (containing the custom OS definitions) before building the image. 
I can't find any trick to do that properly.

guysoft, 2020-12-01:
Not really.
The mount code is here and its mostly hard-coded in aspect of what goes where:
https://github.com/guysoft/CustomPiOS/blob/devel/src/common.sh#L162

If you can share what your code is doing exactly it might help.

Salamandar, 2020-12-01:
I understand.
If there was a before_chroot_script I could do without a bind mount.
For now, I have an acceptable solution : I edited the build_dist script : 
```
#!/usr/bin/env bash
DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"

export DIST_PATH=${DIR}
export CUSTOM_PI_OS_PATH="$DIR/../CustomPiOS/src"
export PATH=$PATH:$CUSTOM_PI_OS_PATH

rsync -av --delete \
    "$DIR/../../StrangerFamily" \
    "$DIR/modules/strangerpi/filesystem/home/pi"

"$CUSTOM_PI_OS_PATH/build_custom_os" "$@"
```

guysoft, 2020-12-01:
That should not be too hard to add.
It could be added here:
https://github.com/guysoft/CustomPiOS/blob/devel/src/custompios#L147
It will have to be though one script for the whole distro and not one per-module. I think it would make more sense that way.

umlaeute, 2023-09-14:
i guess this should have read https://github.com/guysoft/CustomPiOS/blob/333dd9f46b84e5b80104e5163d9c2574b4579fef/src/custompios#L147
(clicking on the ellipsis besides a line of code, the pull down menu has an entry "Copy permalink" which will make the reference usable even when the line-numbers change)