guysoft/CustomPiOS #93: BASE_RELEASE_IMG_NAME & BASE_RELEASE_ZIP_NAME not working as expected.

KwadFan, 2020-12-17:
After fideling around with those Variables, I found out that the Handling in `release` script is not working like the Vars mentioned to be.

So, I wrote a simple test script to check what should be done and what is done by default.
I figured out that you leave old vars or simply set some typos :)

After I figured out how to manipulate a docker Image i could test my result und it works like I would expect it should work.

Now you can set those Names in config file and the Image would be renamed ( as I expect ).

Here my test script:
`#!/bin/bash
set -x

[ -n "$BASE_RELEASE_IMG_NAME" ] || BASE_RELEASE_IMG_NAME=default
[ -n "$BASE_RELEASE_ZIP_NAME" ] || BASE_RELEASE_ZIP_NAME=default

DIST_NAME=ImageNameHandlingTEST
DIST_VERSION=0.0.1-alpha

BASE_DISTRO=raspian
BASE_RELEASE_COMPRESS=yes
BASE_RELEASE_IMG_NAME=myimagename
#BASE_RELEASE_ZIP_NAME=myzipname.zip
#BASE_RELEASE_IMG_NAME=${DIST_NAME}-${DIST_VERSION}
#BASE_RELEASE_ZIP_NAME=myzipname-${DIST_VERSION}
FILENAME=2020-12-02-raspios-buster-armhf-lite.img

if [ "${BASE_RELEASE_IMG_NAME}" == "default" ]; then
    if [ "${BASE_DISTRO}" == "ubuntu" ]; then
        CHANGE_STRING_LIST=( "ubuntu" )
    else
        CHANGE_STRING_LIST=( "raspbian" "raspios")
    fi

    # Handle variant name
    for CHANGE_STRING in "${CHANGE_STRING_LIST[@]}"
    do
        if [ -z "$1" ] || [ "$BASE_IGNORE_VARIANT_NAME" == "yes" ] ;then
            IMG_FILENAME=$(echo "${FILENAME::-4}"-"${DIST_VERSION}" | sed "s/${CHANGE_STRING}/${DIST_NAME,,}/").img
        else
            IMG_FILENAME=$(echo "${FILENAME::-4}"-"${DIST_VERSION}" | sed "s/${CHANGE_STRING}/${DIST_NAME,,}-$1/").img
        fi
    done
else
    # Added little cleanout, if someone is stupid like me and adds .img :)
    IMG_FILENAME="${BASE_RELEASE_IMG_NAME%.img}.img"
fi

echo -e "\e[1;31m" -e "${FILENAME}" -e "${IMG_FILENAME}" -e "\e[0m"

if [ "${BASE_RELEASE_COMPRESS}" == "yes" ]; then
    echo "Compressing image"
    # Default behavior if no names set 
    if [ "${BASE_RELEASE_IMG_NAME}" == "default" ]; then
        echo -e "${IMG_FILENAME::-4}.zip" -e "${IMG_FILENAME}"
    else
        # If Zip name not given, rename to given img name
        if [ "${BASE_RELEASE_ZIP_NAME}" == "default" ]; then
            echo -e "${BASE_RELEASE_IMG_NAME%.img}".zip -e "${IMG_FILENAME}"
        else
            echo -e "${BASE_RELEASE_ZIP_NAME%.zip}".zip -e "${IMG_FILENAME}"
        fi
    fi
else
    echo "Not compressing because setting is disabled"
fi`

according to that I'll offer a PR.

Regards Stephan

KwadFan, 2020-12-17:
Here the output from my build.log:

`+++ compgen -A variable PASSWORD_FOR_SUDO_                                                                                                                                         [1209/1880]
++ echo ================================================================                                                                                          
================================================================                             
+++ ls .                                                                                 
+++ grep -e '.img$' -e '.raw$'                                                           
+++ tail -n 1                                                                                                              
++ basename 2020-12-02-raspios-buster-armhf-lite.img                                                                                                                                         
+ FILENAME=2020-12-02-raspios-buster-armhf-lite.img                                                                                                                
+ '[' hahah == default ']'                                                                                                                    
+ IMG_FILENAME=hahah.img                                                           
+ mv 2020-12-02-raspios-buster-armhf-lite.img hahah.img               
+ '[' yes == yes ']'                                                                                          
+ echo 'Compressing image'                                                                                               
Compressing image                                                                                   
+ '[' hahah == default ']'                                                                    
+ '[' hahah == default ']'                                                                                                                                                        
+ zip hahah.zip hahah.img                                                                                   
  adding: hahah.img (deflated 88%)                                                       
+ popd                                                         
/distro                                                                                                                                                                    
+ chmod 777 /distro/workspace/aptcache /distro/workspace/chroot_script /distro/workspace/hahah.img /distro/workspace/hahah.zip /distro/workspace/mount          
+ popd                                                                                   
/                                                                                                    
docker-compose down                                                                                                                                                                          
Stopping fullRigg3DPI-build ... done                                                                   
Stopping custompios-watcher ... done                                                               
Removing fullRigg3DPI-build ... done                                                    
Removing custompios-watcher ... done                                                                     
Removing network fullrigg3dpi_default                                                                  
kwadfan@debian-vm:~/devel/Pi/fullRigg3DPI$ ls                                                      
build.log  docker-compose.yml  fullRigg3DPI.code-workspace  Makefile  run.sh  src                                                       
kwadfan@debian-vm:~/devel/Pi/fullRigg3DPI$ cd src/                                                                                                               
kwadfan@debian-vm:~/devel/Pi/fullRigg3DPI/src$ cd workspace/                                                                   
kwadfan@debian-vm:~/devel/Pi/fullRigg3DPI/src/workspace$ ls                                        
aptcache  chroot_script  hahah.img  hahah.zip  mount        `

guysoft, 2020-12-24:
Closed via #94

closed 2020-12-24 by guysoft