guysoft/CustomPiOS #101: Unpack function fails when building against current Raspios image following upstream /usr merge

SketchyFletchy, 2021-01-14:
Upstream Debian has moved to unify `/lib`, `/bin` and `/sbin` into `/usr/…` for various reasons outlined here: https://www.freedesktop.org/wiki/Software/systemd/TheCaseForTheUsrMerge/

This change made it’s way into the raspios builder Pi-Gen some time ago and has now filtered down into official releases. See https://github.com/RPi-Distro/pi-gen/issues/396

When building a new distro with the latest Raspios release the builder will fail when attempting to run the `unpack` function targeting the root folder of the new destination if the content includes files in /lib, /bin or /sbin as the copy function reports it can’t write to the target location as it is a file (symlink) rather than a directory.

I’ll have a look at what can be done to better direct the copy function in `unpack` but for the moment the best workaround I’ve found is to use a late-2020 Raspios release prior to the usr merge. At time of writing there were no major security updates included in the latest release that aren’t present in prior releases. Additionally, new modules writing content to these locations should probably migrate to placing content into /usr/… for future compatibility.

guysoft, 2021-01-18:
Hey,
I already fixed this in the commit here: https://github.com/guysoft/OctoPi/commit/8aa44187aafa52e487f2b6661007a5aad35c62a6
Which was reported here: https://github.com/guysoft/OctoPi/issues/703 .
This is in specifically in the OctoPi distro when unpacking to that location by this. It was due to filesystem path not being updated.
I didn't see any module here affected by it in CustomPiOS. 

What are you trying to build in particular?

guysoft, 2021-02-11:
Closing if no reply, since this seems to be fixed

SketchyFletchy, 2021-05-18:
Sorry for super slow reply! Happy to confirm it’s fixed, I just had to update local CustomPiOS scripts. My bad.

Thanks!

closed 2021-05-18 by SketchyFletchy