guysoft/CustomPiOS #104: Change default 'pi' password?

grantemsley, 2021-01-20:
Can the existing admintools module be used to change the password for the pi user? Or does that functionality already exist in one of the modules?

It looks like you have to specify a new user to be created if you want to set a password.

grantemsley, 2021-01-20:
I haven't tested it, but I think if you changed the 'useradd' lines to continue on error it would let you specify `ADMIN_TOOLKIT_NAME=pi` and change the password and add SSH keys for the existing user.

Either that, or if you don't like ignoring potential errors, check if the account exists and skip creating it if it does.

guysoft, 2021-01-20:
There is no functionality for that, it would need to be added. Probably in the base module. 
Currently it only sets the password if the user does not exist:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/base/start_chroot_script#L43

PR #105 (2021-01-24): Allow altering pi account
Use id to see if the account specified already exists. If it does, skip creating it but still change the password and setup SSH keys.
This allows the module to be used to secure the pi account without having to create a new account.

Tested on one of my builds, and it changed the password and added my SSH keys perfectly.  This should only change the behaviour if you specify `ADMIN_TOOLKITNAME=pi`, which previously would cause the build to just die at the `useradd`, so it shouldn't affect any existing builds using the module.

Fixes #104
closed 2021-01-24 by guysoft