guysoft/CustomPiOS #106: aptcache folder in worksapce after build

masseelch, 2021-02-09:
When building with docker `docker run --privileged -t -v "$SCRIPTPATH/src":/distro --device /dev/loop-control guysoft/custompios:devel build` i always end up with the attached workspace tree.

[image]

guysoft, 2021-02-10:
Mounting happens here:
https://github.com/guysoft/CustomPiOS/blob/devel/src/custompios#L120

You could set ``export BASE_APT_CACHE=no`` in your config file. That way it would not mount at all.

As to figuring out why this is happening - are you using the docker build method or something else?

masseelch, 2021-02-10:
Mounting the cache is to fasten up subsequent builds i guess? I'd like to keep it then. 

I will set `export BASE_APT_CACHE=no` and run a build. 

I use docker to build.
> When building with docker docker run --privileged -t -v "$SCRIPTPATH/src":/distro --device /dev/loop-control guysoft/custompios:devel build i always end up with the attached workspace tree.

masseelch, 2021-02-10:
> I see you are using docker. Unsure then - what's the host OS and Docker version? It might be something as to how its mounting files.

```shell
» lsb_release -a
LSB Version:	n/a
Distributor ID:	ManjaroLinux
Description:	Manjaro Linux
Release:	20.2.1
Codename:	Nibia
```
```shell
» docker --version
Docker version 20.10.2, build 2291f610ae
```

> I will set export BASE_APT_CACHE=no and run a build.

This "fixed" the issue.

guysoft, 2021-02-10:
I will try giving a bunch of guesses because I don't know what is causing it in the system. Perhaps one of them might be it.
* the build log might have some "no such file or directly" error, similar to [here](https://unix.stackexchange.com/questions/566494/unable-to-bind-mount-in-arch-linux)
* The filesystem is not ext4 and its causing something strange
* You likely have, but just for good measure - try stopping/starting the container
* I need to test that Docker version
* Might be an Arch thing, I know the kernel might be a little different there for that kind of stuff

masseelch, 2021-02-10:
I did several succesful builds of FullPageOS and as far as i can tell the `BASE_APT_CACHE=no` is not set there. 

> - the build log might have some "no such file or directly" error, similar to here

I dont see such error. Build log is attached.

> - The filesystem is not ext4 and its causing something strange

```shell
» mount | grep /dev/nvme
/dev/nvme0n1p5 on / type ext4 (rw,noatime)
/dev/nvme0n1p1 on /boot/efi type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,utf8,errors=remount-ro)
```
> - You likely have, but just for good measure - try stopping/starting the container

The container does not run in detached mod. Its started on every build and shuts down after. I did prune images and container, repulled `guysoft/custompios:devel`, but error remains.

> - Might be an Arch thing, I know the kernel might be a little different there for that kind of stuff

I run the `5.4.89-1-MANJARO` Kernel. I did try a build with the `4.19`, which is the one used by Buster i think. Still the same.

The only real difference is, that this build does use the raspbain (raspberry pi os) WITH Desktop, while fullpageos does use the `lite` variant. 

Id you need more info i am happy to provide.

guysoft, 2021-02-10:
I could try and reproduce using the desktop version, but not sure I have the time for that. Leaving open for now if I do.

masseelch, 2021-02-10:
Building FullPageOS with the same image does result in the same error. Folder `src/workspace/aptcache/archives` contains `partial` having the ownership `103:root`.