guysoft/CustomPiOS #107: Question: Is it possible to use and .img instead of a zipped one

masseelch, 2021-02-11:
Looks like it is not. Would be a nice feature though. I do a lot of builds and they could be fastened up a lot if the image decompressing step on start would be skipped when providing an already unzipped image.

I can provide (or at least try to) a contribution.

masseelch, 2021-02-11:
I see there is this in `custompios`:

```shell
  if [[ $BASE_ZIP_IMG =~ \.img$ ]]; then
    # if the image is already extracted copy over
    cp "$BASE_ZIP_IMG" .
  else
    7za x -aoa "$BASE_ZIP_IMG"
  fi
```

But i am not able to give the correct path to my extracted image. I always get `Error: could not find image: ....`

masseelch, 2021-02-11:
Got it!

For everyone landing here: The `custompios` does its magic in the `workspace` folder. Adding this to my config:
```bash
export BASE_ZIP_IMG=../image/2021-01-11-raspios-buster-armhf.img
```
made it work.

closed 2021-02-11 by masseelch

guysoft, 2021-02-11:
Thanks for reporting, if you think this should go anywhere in the documentation then you are welcome to put in the wiki or wherever you were searching the the solution.

masseelch, 2021-02-11:
Might be a good idea to put a comment into [config](https://github.com/guysoft/CustomPiOS/blob/43cf550082de321ec5ba027dce49297bcdff5f09/src/modules/base/config#L17).