guysoft/CustomPiOS #109: Clarification on build process

pablosproject, 2021-02-11:
Hey, thanks for the work you are doing, this tool is great. 
I am trying to think about how to adapt my script to create my custom Pi application using CustomPiOS. What happens if a module needs a reboot?
As an example I have a script that does some preparation, I run it and reboot (I am increasing the GPU memory which needs a reboot). After the reboot, I run the build script which configures the Pi to the state that I want.

Is it something that I can accomplish with CustomPiOS?

guysoft, 2021-02-14:
Hey,
CustomPIOS does not run exactly run the kernel. Its using qemu to emulate running ARM instructions. 
If you are changing the GPU memory for when the Pi boots, its not necessary to reboot during the build process. Because you are not actually running the Rpi kernel, only emulating it. There is no GPU memory emulated at all.
The GPU memory change would be relevant when you boot the built image on a physical Pi.

Just to give an example of a case where you do need to reboot - when the image starts for the first time it resizes the disk partition. So what is done in that case, is you have a startup script that on first boot resizes the image, then reboots. Then all that happens on the physical Pi with its SD card.

pablosproject, 2021-02-15:
Thanks for your reply, cristal clear. Keep up the good work. Cheers.

closed 2021-02-15 by pablosproject