guysoft/CustomPiOS #111: Performances using docker

pablosproject, 2021-02-16:
Hi, sorry for opening an issue, but I couldn't find a better way to ask questions (maybe a discord community?).

I was digging into using docker for building my custom image, and everything is working fine.
Now the tricky part: the build on my Macbook 2015 is taking around 30 minutes (more or less); I also need to build 3 variants of my image. The build process is not doing anything super fancy: update packages, compile a C++ executable (compilation takes 5 minutes on the pi3 itself), and then registers it as a starting service.

I'm thinking of running a nightly build and automatically generate those images at every commit and changes to my repo, but this can easily consume all the minutes of my GitHub actions.
Also testing the build script locally on my Mac is being painful, as it's super slow. 

Do you have any suggestions on how to improve the compilation time? So far I tried:

- Running it in an Ubuntu Virtual machine (failing I think because QEMU does not run properly on VM)
- Running it on the pi (performances ok, but I need a second pi after that to test that the image works properly)

Thanks for your time, I really appreciate it.

guysoft, 2021-02-16:
1. Running it on the Pi would be the fastest option. In the sense that you already have the code. It will run faster because you are not going though emulation with qemu. That is what I did for RealtimePi, and it made the compile time go from 12 hours to 2. Source: https://github.com/guysoft/RealtimePi/issues/1
2. You could also cross-compile the binary on amd64 and then just copy it to the Pi image. That way you are not emulating a compiler which is much more heavy. Depends what toolchain you are using. If you are also building a few variants that use the same compiled code that way it will be the same code across variants.

I don't see a reason to run it on an Ubuntu Virtual VM. You need virtualization, not emulation.

pablosproject, 2021-02-16:
I tried Ubuntu as I had VirtualBox installed on my mac and wanted to check if performances were better by running it there rather than on the docker image.

Very clear explanation, thanks a lot.

closed 2021-02-16 by pablosproject