guysoft/CustomPiOS #113: Wrong user when running raspi-config

pablosproject, 2021-02-18:
I just want to report that using `raspi-config` inside a chroot script may lead to unexpected behaviour. Here is my case:

``` bash
#!/usr/bin/env bash
########

# Source error handling, leave this in place
set -x
set -e

source /common.sh
install_cleanup_trap

unpack /filesystem/home/pi /home/"${BASE_USER}" "${BASE_USER}"
unpack /filesystem/home/root /root root
unpack /filesystem/boot /boot

# -------------------------------------------------------
#           DEPENDENCIES FOR INSTALLATION
# -------------------------------------------------------
apt-get update && apt-get -y upgrade

apt-get install -y libsdl2-dev .....

# -------------------------------------------------------
#           Binary compilation
# -------------------------------------------------------
...

# Unpack root at the end, so files written in /etc... are not overwritten
unpack /filesystem/root /

# Make emulation binary executable by current user
chmod +x /usr/bin/compiled

# Configure auto login as pi user (using raspi-config)
raspi-config nonint do_boot_behaviour B2
sudo systemctl set-default graphical.target

``` 

The last 2 lines, is where things went wrong. Basically once you ran raspi-config the autostart of getty (`/etc/systemd/system/getty@tty1.service.d/autologin.conf`) contains the following lines:
``` bash
[Service]
ExecStart=
ExecStart=-/sbin/agetty --autologin  --noclear %I $TERM
```
Instead of
``` bash
[Service]
ExecStart=
ExecStart=-/sbin/agetty --autologin pi --noclear %I $TERM

```

Not sure if we should change the user with which we are running the chroot script or if it's a raspi-config problem, but I just spent half a day with it and want to report in case someone else is using the same approach to make the user auto-login.

guysoft, 2021-05-31:
Thanks for updating. Sorry missed this.

It might be a good idea to open an issue at: https://github.com/RPi-Distro/raspi-config/issues

In FullPageOS I took a look in the source to see how to switch on the GPU.

guysoft, 2021-05-31:
Reported here: https://github.com/RPi-Distro/raspi-config/issues/160

closed 2022-10-21 by pablosproject