guysoft/CustomPiOS #114: Qemu-arm-static busy with 2x nested modules.

amrsoll, 2021-02-26:
The module list that I have is the following 
```sh
export MODULES="base(raspicam, network, disable-services(dev-libs(octopi, beamos)), password-for-sudo)"
```

It would seem that CustomPiOS tries to copy the `qemu-arm-static` file while it's already being used, possibly because it should only be copied once instead of doing copied for each nested module?

```
+ execute_chroot_script /home/runner/work/BeamOS/BeamOS/repository/src/modules/dev-libs /home/runner/work/BeamOS/BeamOS/repository/src/modules/dev-libs/end_chroot_script
+ '[' -f /.dockerenv ']'
+ '[' -d /home/runner/work/BeamOS/BeamOS/repository/src/modules/dev-libs/filesystem ']'
++ arch
+ '[' x86_64 '!=' armv7l ']'
++ arch
+ '[' x86_64 '!=' aarch64 ']'
+ '[' armv7l == armv7l ']'
++ which qemu-arm-static
+ cp /usr/bin/qemu-arm-static usr/bin/qemu-arm-static
cp: cannot create regular file 'usr/bin/qemu-arm-static': Text file busy
++++ echo_red 'build failed, unmounting image...'
```

N.B. It would also seem that my `dev-libs/end_chroot_script` is not actually run as I can't see the output. Do you want me to open a separate ticket?

See [Build log](https://github.com/mrbeam/BeamOS/runs/1986261529)

amrsoll, 2021-02-26:
The purpose of the `dev-libs` module is to take care of `apt` packages - it updates `apt` and installs all of the required source libraries and removes them at the end of the build so they aren't present in the final image.

guysoft, 2021-03-01:
Between modules loads qemu-arm-static should not be used, and therefore should not be busy.

Does #116 fix this issue, or tells you which process is still running?

guysoft, 2024-03-23:
Closing if no reply

amrsoll, 2024-04-01:
I can't check any more, I'll close it :)

closed 2024-04-01 by amrsoll