guysoft/CustomPiOS #118: losetup -d seems to be called twice, resulting in an error

felurx, 2021-03-07:
At the end (?) of running build_dist I get the following error/output:
```
+++ e2fminsize_blocks=2906793
+++ e2ftarget_mb=1619
+++ e2ftarget_blocks=3316393
+++ e2fsize_mb=1903
+++ size_offset_mb=284
+++ losetup -d /dev/loop0
+++ echo 'Actual size is 1903 MB (3899392 blocks), Minimum size is 1419 MB (363349 file system blocks, 2906793 blocks)'
Actual size is 1903 MB (3899392 blocks), Minimum size is 1419 MB (363349 file system blocks, 2906793 blocks)
+++ echo 'Resizing to 1619 MB (3316393 blocks)' 
Resizing to 1619 MB (3316393 blocks)
+++ '[' 284 -gt 0 ']'
+++ echo 'Partition size is bigger then the desired size, shrinking'
Partition size is bigger then the desired size, shrinking
+++ shrink_ext 2020-02-13-raspbian-buster-lite.img 2 1618
+++ image=2020-02-13-raspbian-buster-lite.img
+++ partition=2
+++ size=1618
+++ echo 'Resizing file system to 1618 MB...'
Resizing file system to 1618 MB...
++++ grep 2020-02-13-raspbian-buster-lite.img2
++++ awk '{print $4-0}'
++++ sfdisk -d 2020-02-13-raspbian-buster-lite.img
+++ start=532480
+++ offset=272629760
+++ detach_all_loopback 2020-02-13-raspbian-buster-lite.img
++++ losetup
++++ grep 2020-02-13-raspbian-buster-lite.img
++++ awk '{ print $1 }'
+++ for img in $(losetup  | grep $1 | awk '{ print $1 }' )
+++ losetup -d /dev/loop0
losetup: /dev/loop0: detach failed: No such device or address
+ exit 1
```
Apparently, `losetup -d` is being called twice, and the second time it doesn't have anything to detach, so it fails and the script dies. What could I be doing wrong here?

I have tried doing `sudo modprobe loop max_loop=128`, but that didn't seem to change anything.
[Here's my build.log](https://github.com/guysoft/CustomPiOS/files/6097257/build.log)

Is there anything else I should provide to make narrowing this down easier?

andysheen, 2021-03-15:
@felurx I had this issue with project I was using buster lite and CustomPiOs. Mine seemed to be fixed by using the develop branch instead. I could be wrong, but I think this may have fixed it https://github.com/guysoft/CustomPiOS/commit/4e4c7f8105026facfecf9c5f2ab5541d24283d54

guysoft, 2021-03-15:
Would be helpful to confirm is that solves the issue.

andysheen, 2021-03-16:
hi @guysoft . First off, thanks for all the work on CustomPiOS! I just set up three builds to test the fix. One with your develop branch, one with the master branch and one with a fork of your master branch with https://github.com/guysoft/CustomPiOS/commit/4e4c7f8105026facfecf9c5f2ab5541d24283d54 added to it. The only Action that passed was the develop branch, So i guess it is not the commit I thought fixing it!

Here's the builds if you're interested:

using the develop branch ✅
https://github.com/andysheen/NaturewatchCameraServer/actions/runs/657193221

using the master branch ❌
https://github.com/andysheen/NaturewatchCameraServer/actions/runs/657198902

using a forked branch of master + losetup fix ❌
https://github.com/andysheen/NaturewatchCameraServer/actions/runs/657227453

felurx, 2021-03-16:
@andysheen Thanks a lot for the hint! I will try the develop branch this evening.

felurx, 2021-03-18:
Okay, so I seem to have stretched the definition of "this evening" a bit... (Well, actually, I completely forgot about it. Sorry about that :frowning_face: )
It seems that I have already been on the `devel` branch, which confused the hell out of me, but I just had to do a quick `git pull`, and after that everything seems to have worked perfectly! (Or, at the very least, the build completed :P)
Thank you very much!

closed 2021-03-18 by felurx