guysoft/CustomPiOS #120: Docker build: Could NOT find OpenGLES

pablosproject, 2021-03-18:
Hi there, I was using docker to build my images and run into this issue. Is there something that got changed lately in the Docker image that could have affected either OpenGLES or OpenGLv2? 
Thanks

pablosproject, 2021-03-22:
By looking deeper at this issue I saw that it is related to compiling my binary using CMake. Looks like for some reason libraries paths are not found (and thus I changed the title of this issue).

I see that there are other distros affected by this issue, like [FluiddPi](https://github.com/cadriel/FluiddPI/runs/2162007404?check_suite_focus=true). As you can see the build image is broken on the CMake steps due to missing libraries (cc @cadriel ).

Do you have any hint from where I shall start looking for this issue? I'll be happy to give a hand but I know very little about this and I need help for starting out.
I already checked the history of all docker files and found that there is nothing wrong, so now I will start looking at the image generation code.

guysoft, 2021-03-25:
There is this talk+PR on OctoPi:
https://github.com/guysoft/OctoPi/pull/731

If you want to try and use that version of came, that might help me narrow down if that version of cmake solves the issue.

pablosproject, 2021-03-26:
Hey, I did tried using `apt-get install cmake=3.13.4-1 cmake-data=3.13.4-1` and now everything build fine again. I'll use that for now on in my code, let me know If you need something else from me to test.

guysoft, 2021-03-26:
Ok, thanks for updating, I will merge it here too.

guysoft, 2021-04-08:
Looks like this also fixed the build on github actions here. so closing.

closed 2021-04-08 by guysoft