guysoft/CustomPiOS #124: the java command requires a mounted proc fs (/proc).

pablosproject, 2021-04-08:
Hi, I tried installing open-jdk in my install `start_chroot_script` using `apt-get install -y default-jdk`, and I am getting the following error:

```
the java command requires a mounted proc fs (/proc).
dpkg: error processing package openjdk-11-jre-headless:armhf (--configure):
 installed openjdk-11-jre-headless:armhf package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of openjdk-11-jre:armhf:
 openjdk-11-jre:armhf depends on openjdk-11-jre-headless (= 11.0.9.1+1-1~deb10u2); however:
  Package openjdk-11-jre-headless:armhf is not configured yet.
```

Any idea where I should start looking to solve the issue?

guysoft, 2021-04-08:
Hey,
Looks like this might require adding code to mount ``/proc``.

There is a solution suggested here:
https://stackoverflow.com/a/44572431/311268

The lines in CustomPiOS needed to change should go here:
https://github.com/guysoft/CustomPiOS/blob/devel/src/common.sh#L191

And something might need to go here:
https://github.com/guysoft/CustomPiOS/blob/devel/src/common.sh#L196

Do you think you can manage testing this?

pablosproject, 2021-04-08:
I'll try get a minimal error out and try reproducing it, and then work on the solution, seems straightforward. Will work on this as soon that I have some spare time.

guysoft, 2021-04-11:
Pushed a fix, please ping me if that solves it

closed 2021-04-11 by guysoft

pablosproject, 2021-04-12:
Solved, working correctly. Thanks.

commit c61b80e5 (2021-04-11): Add mounting of /proc . Fixes #124