guysoft/CustomPiOS #125: /dev/null Permission denied

pablosproject, 2021-04-13:
Hi there. After the fix here https://github.com/guysoft/CustomPiOS/commit/c61b80e5dc27cf98daab55abfc6a77ac30d381da everything works fine, but there are now issues with apt (I tried with a clean install and on the two projects I am working on).

To reproduce it simply try to launch `apt-get update` with the latest docker image either inside the docker image using `docker exec -it /bin/bash -c "sudo apt install"`, or by putting it inside a script.. That command outputs:
``` bash
+ apt-get update
Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.9 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
gpgv: Signature made Tue Apr 13 08:01:19 2021 BST
gpgv:                using RSA key CF8A1AF502A2AA2D763BAE7E82B129927FA3303E
gpgv: Good signature from "Raspberry Pi Archive Signing Key"
/usr/bin/apt-key: 57: /usr/bin/apt-key: cannot create /dev/null: Permission denied
Get:3 http://archive.raspberrypi.org/debian buster/main armhf Packages [372 kB]
gpgv: Signature made Tue Apr 13 06:08:40 2021 BST
gpgv:                using RSA key A0DA38D0D76E8B5D638872819165938D90FDDD2E
gpgv: Good signature from "Mike Thompson (Raspberry Pi Debian armhf ARMv6+VFP) <mpthompson@gmail.com>"
/usr/bin/apt-key: 57: /usr/bin/apt-key: cannot create /dev/null: Permission denied
Get:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [104 kB]
```

To solve that issue I've put in my start_chroot script the following
``` bash
    rm /dev/null
    mknod /dev/null c 1 3
    chmod 666 /dev/null
```
which now makes everything work, but at the end of the compilation it gets an issue when unmounting /dev

```
+++ sudo umount /distro/workspace/mount/dev
umount: /distro/workspace/mount/dev: target is busy.
+ exit 1
```

Is there any options to the mounting point in docker that can help us solve the permission issue on /dev/null?

guysoft, 2021-04-13:
Try recreating your container. Tested and apt works on docker here

pablosproject, 2021-04-13:
I did recreate the container, but the issue is still present. I am using `guysoft/custompios:develop` as my main container.

It's strange, because if I run it as a standalone container with:
```
docker run -it  guysoft/custompios:devel /bin/bash
root@871c9838852c:/# apt update
```
everything works fine.

When mounting a volume, the issue is happening, here my docker-compose, nothing fancy:
```
version: "3.6"

services:
  custompios:
    image: guysoft/custompios:devel
    container_name: xxx-kiosk-create
    tty: true
    restart: always
    privileged: true
    volumes:
      - ./xxx/src:/distro
    devices:
      - /dev/loop-control

```

pablosproject, 2021-04-13:
On the left, I am running inside the docker-compose mounted container, on the right using `docker run -it  guysoft/custompios:devel /bin/bash`.

Docker version 20.10.5, build 55c4c88
[image]

pablosproject, 2021-04-13:
Apart from the issue resolved, I solved in my use case doing in the `start_chroot_script`:
```
    rm /dev/null
    mknod /dev/null c 1 3
    chmod 666 /dev/null
```
And in the `end_chroot_script` doing: `fuser -cuk /dev` to release the resource and correctly unmount. Let me know if you need more help for the docker part.

guysoft, 2021-04-13:
Help is welcome, but I am not sure what is causing it to be busy.
It might be something you are running in your build, which I can't see

pablosproject, 2021-04-13:
That issue happens in various build project. The screenshot I sent before is just after creating the container with docker composer, the build was never launched.

Anyway, I have a solution for it and it works for me, will check if maybe is docker by trying on another machine.

Feel free to close it, or keep it as a reminder. Thanks for the help.

closed 2021-05-31 by guysoft