guysoft/CustomPiOS #129: Using CustomPiOS with gitlab CI/CD

steviehs, 2021-05-27:
I want to build using gitlabs CI/CD (the hosted one as it is a FOSS project), as far as I understand, the image there are not privileged, so I do not get any loopback support.
Any hints in still able to work with it? A final thought would be to have a full blown debian running in a qemu, but it really feels like overkill...

guysoft, 2021-05-27:
There is a github action script. I think it might be adapted to work for gitlab. But I am not sure how:
https://github.com/guysoft/OctoPi/blob/devel/.github/workflows/build.yml

steviehs, 2021-05-27:
yep, basically it is adaptable, but: not for gitlab's public runners, as they are not privileged. Right now we are using our own runner, but I would be great to go without it. But I do not really see a good way except from my above proposed one.

guysoft, 2021-05-27:
That would be great - however I am not sure how to mount an image without loopback. So I am affraid I have no idea how to solve this issue without gitlab.com changing something on their end.
You could set up your own runner or gitlab server.

steviehs, 2021-08-19:
We have done it this way. If anybody is interested to use CustomPiOS with gitlab CI/CD, have a look at https://gitlab.com/syncosync/syncosync/
Thanks for your support!

closed 2021-08-19 by steviehs

guysoft, 2021-08-19:
@steviehs that project is really cool! I was trying to code something similar myself with [BackupFriendPi](https://github.com/guysoft/BackupFriendPi) and [backupfriend-client](https://github.com/guysoft/backupfriend-client).
Would you like me to add syncosync to this list?
https://github.com/guysoft/CustomPiOS#list-of-distributions-using-custompios
(Or you could PR yourself) .