guysoft/CustomPiOS #130: Issues with `pm2`

hrueger, 2021-06-16:
Hi @guysoft,
First of all, thanks for this great project.
I want to use pm2 to keep my node script running. So I built a module and its `start_chroot_script` includes the following:
```sh
# Add MyScript to the startup
sudo npm install -g pm2
cd /home/pi/MyScript
pm2 start headless.js
pm2 startup systemd
sudo env PATH=$PATH:/usr/bin /usr/lib/node_modules/pm2/bin/pm2 startup systemd -u pi --hp /home/pi
```
This part works fine, but at the logs end like this:
```+++ '[' -n '' ']'
+++ '[' -n '' ']'
+++ '[' yes == yes ']'
+++ restoreLd
+++ '[' -f etc/ld.so.preload ']'
+++ sed -i 's@\#/usr/lib/arm-linux-gnueabihf/libcofi_rpi.so@/usr/lib/arm-linux-gnueabihf/libcofi_rpi.so@' etc/ld.so.preload
+++ sed -i 's@\#/usr/lib/arm-linux-gnueabihf/libarmmem.so@/usr/lib/arm-linux-gnueabihf/libarmmem.so@' etc/ld.so.preload
+++ sed -i 's@#/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so@/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so@' etc/ld.so.preload
+++ popd
/distro/workspace /distro
+++ unmount_image /distro/workspace/mount
+++ mount_path=/distro/workspace/mount
+++ force=
+++ '[' 1 -gt 1 ']'
+++ sync
+++ '[' -n '' ']'
++++ sudo mount
++++ grep /distro/workspace/mount
++++ awk -F ' on ' '{print $2}'
++++ awk '{print $1}'
++++ sort -r
+++ for m in $(sudo mount | grep $mount_path | awk -F " on " '{print $2}' | awk '{print $1}' | sort -r)
+++ echo 'Unmounting /distro/workspace/mount/var/cache/apt...'
Unmounting /distro/workspace/mount/var/cache/apt...
+++ sudo umount /distro/workspace/mount/var/cache/apt
+++ for m in $(sudo mount | grep $mount_path | awk -F " on " '{print $2}' | awk '{print $1}' | sort -r)
+++ echo 'Unmounting /distro/workspace/mount/proc...'
Unmounting /distro/workspace/mount/proc...
+++ sudo umount /distro/workspace/mount/proc
umount: /distro/workspace/mount/proc: target is busy.
+ exit 1
```
The Image file is about 2,8 GB (without installing and setting up pm2 like described above it is only about 500 MB) and it is also corrupted.

pablosproject, 2021-06-16:
Hey, I happened to see this issue and I can help solving some doubt. First: the image size depend on which image you are basing your original one against, so he size you are obtaining seems fine.
Did you built with docker?

Do you mind sharing the script so we can try to replicate the issue?

hrueger, 2021-06-16:
Wow, that was a quick response!
Yes, I'm using docker on windows to build for armv7 (Raspberry Pi). My Base Image is the Raspberry Pi OS Lite Image 2021-05-07.
I'll create a reproduction repo.

hrueger, 2021-06-16:
here's my test-repo: https://github.com/hrueger/custompios-pm2-issue

pablosproject, 2021-06-17:
Ok, so the issue is that `/proc` is used and cannot be unmounted, so the build process is not complete, that's where the issues you are experiencing come from.

Looking at who is using `/proc`, here is what I got running `lsof | grep /proc`:
```
PM2\x20v5 2791                root   24r      REG              0,208        0   35431 /distro/workspace/mount/proc/2804/stat
PM2\x20v5 2791 2793 node      root   24r      REG              0,208        0   35431 /distro/workspace/mount/proc/2804/stat
PM2\x20v5 2791 2794 node      root   24r      REG              0,208        0   35431 /distro/workspace/mount/proc/2804/stat
PM2\x20v5 2791 2795 node      root   24r      REG              0,208        0   35431 /distro/workspace/mount/proc/2804/stat
PM2\x20v5 2791 2796 node      root   24r      REG              0,208        0   35431 /distro/workspace/mount/proc/2804/stat
PM2\x20v5 2791 2797 node      root   24r      REG              0,208        0   35431 /distro/workspace/mount/proc/2804/stat
PM2\x20v5 2791 2798 node      root   24r      REG              0,208        0   35431 /distro/workspace/mount/proc/2804/stat
PM2\x20v5 2791 2799 node      root   24r      REG              0,208        0   35431 /distro/workspace/mount/proc/2804/stat
PM2\x20v5 2791 2800 PM2\x20v5 root   24r      REG              0,208        0   35431 /distro/workspace/mount/proc/2804/stat
PM2\x20v5 2791 2801 PM2\x20v5 root   24r      REG              0,208        0   35431 /distro/workspace/mount/proc/2804/stat
PM2\x20v5 2791 2802 PM2\x20v5 root   24r      REG              0,208        0   35431 /distro/workspace/mount/proc/2804/stat
PM2\x20v5 2791 2803 PM2\x20v5 root   24r      REG              0,208        0   35431 /distro/workspace/mount/proc/2804/stat
lsof      3208                root    3r      DIR              0,208        0       1 /proc
lsof      3208                root    4r      DIR              0,208        0   40054 /proc/3208/fd
```

So, node and PM2 are currently using that directory.
To solve the issue, there are 2 ways:

1. Close/kill all processes that are actually using `/proc`.
2. Force unmount all mounted points. For this I opened this issue https://github.com/guysoft/CustomPiOS/issues/131, let's see what @guysoft thinks about this.

Just put in your end_chroot_script the following lines:
```
# Install fuser
apt-get install -y psmisc

# Kill processes using /proc
fuser -cuk /proc
```
And it will build correctly.

guysoft, 2021-06-17:
Give it a try, should be fixed now

guysoft, 2021-06-17:
@pablosproject Does this run inside docker (hence ``apt-get install -y psmisc`` needs to happen in the docker image, and should be shipped with the docker image). or does it happen inside the image - that is it needs to be shipped on the Pi created image. Because you note to add it in end_chroot_script.

pablosproject, 2021-06-17:
Not sure if it was a question or just correcting what I did. I think this is run inside docker, so that's why we need to stop the process using /proc.

hrueger, 2021-06-17:
>Give it a try, should be fixed now
@guysoft Unfortunately, it does not work. I pulled the latest docker image, but the build logs now end with
```
removed directory 'filesystem'
+++ '[' -n '' ']'
+++ '[' -n '' ']'
+++ '[' yes == yes ']'
+++ restoreLd
+++ '[' -f etc/ld.so.preload ']'
+++ sed -i 's@\#/usr/lib/arm-linux-gnueabihf/libcofi_rpi.so@/usr/lib/arm-linux-gnueabihf/libcofi_rpi.so@' etc/ld.so.preload
+++ sed -i 's@\#/usr/lib/arm-linux-gnueabihf/libarmmem.so@/usr/lib/arm-linux-gnueabihf/libarmmem.so@' etc/ld.so.preload
+++ sed -i 's@#/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so@/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so@' etc/ld.so.preload
+++ popd
/distro/workspace /distro
+++ unmount_image /distro/workspace/mount
+++ mount_path=/distro/workspace/mount
+++ force=
+++ '[' 1 -gt 1 ']'
+++ sync
+++ '[' -n '' ']'
+++ '[' -f /.dockerenv ']'
+++ fuser -cuk /dev
/CustomPiOS/common.sh: line 217: fuser: command not found
+ exit 1
```

pablosproject, 2021-06-17:
You can use `apt-get install -y psmisc `

hrueger, 2021-06-17:
Thanks. Now every user has to do that, right? So maybe it should be added to the docs.

pablosproject, 2021-06-18:
@guysoft what about reverting https://github.com/guysoft/CustomPiOS/commit/63473d5727aa602fc9ad362c53ab1d09f7f52bd7 and specify this issue in the doc?

Otherwise we must force package installation.

hrueger, 2021-06-20:
Unfortunately, it's now not working anymore. I always get
```
+++ sync
+++ '[' -n '' ']'
+++ '[' -f /.dockerenv ']'
+++ fuser -cuk /dev
/CustomPiOS/common.sh: line 217: fuser: command not found
+ exit 1
```
I tried installing `sudo apt-get install -y psmisc`  but it didn't change anything.

guysoft, 2021-06-20:
pushed a new fix just today

hrueger, 2021-06-20:
Yeah, the version I tried was built 16 hours ago.

guysoft, 2021-06-21:
Its still not working,  just checked. I am reverting those too commits. They are breaking builds for everyone.

pablosproject, 2021-06-21:
@hrueger To fix your problem use the snippet I sent you, there is a PR to your Repo with it that builds correctly.

hrueger, 2021-06-21:
Thanks @pablosproject . Now I see why it didn't work when I tried your snippet a few days ago: I installed `psmisc` in the `start_chroot_script` but wanted to use it in the `end_chroot_script`. That's probably the reason why the command was not found.

guysoft, 2021-06-21:
Reverted

hrueger, 2021-07-03:
Thanks a lot for your help. It now works perfectly.

closed 2021-07-03 by hrueger