guysoft/CustomPiOS #131: Force unmount mount points

pablosproject, 2021-06-17:
During the build process, it happened that the resources to unmount at this line were currently busy. https://github.com/guysoft/CustomPiOS/blob/c61b80e5dc27cf98daab55abfc6a77ac30d381da/src/common.sh#L229 

In my builds I solved with the following snippet, killing the process using the busy resource:
```
if [ "$RUN_IN_DOCKER" == "yes" ]; then
    # Release docker retained /dev
    fuser -cuk /dev
fi
```

Would it be harmful to force detach the mount point using `sudo unmount -lf $m`? Or at least catch it and show who is using the mount point in case of failing mount.

guysoft, 2021-06-17:
That would be
```bash
if  [ -f /.dockerenv ]; then
    # Release docker retained /dev
    fuser -cuk /dev
fi
```

closed 2021-06-17 by guysoft

pablosproject, 2021-06-17:
Does this apply to /proc also?

guysoft, 2021-06-17:
@pablosproject I don't know, I don't think so, proc i think is not a normal OS, i think in a way its always busy.

pablosproject, 2021-06-17:
In the related issue the problem was with `/proc` mount being busy. More in general, do you think we can overcome all issues by simply forcing unmount all mounted points?

guysoft, 2021-06-17:
Unmounts fail for a reason.
"Errors should never pass silently.
Unless explicitly silenced."

pablosproject, 2021-06-17:
Makes lot of sense

commit 63473d57 (2021-06-17): Fixes  #131 related #130
commit c890ef4e (2021-06-21): Revert "Fixes  #131 related #130"

This reverts commit 63473d5727aa602fc9ad362c53ab1d09f7f52bd7.