guysoft/CustomPiOS #132: How to pass GitHub workflow variables to CustomPiOS scripts

cooked, 2021-06-28:
Hi,
as per the title I'd like define a variable in the GitHub worflow (and changing with something like strategy.matrix) and pass it down to the CustomPiOS scripts. While I can define the variable no problem, it seems I'm not able to read it properly from one bash script that is part of the filesystem I'm building.
Is it possible? If so, how?

Thanks in advance.

guysoft, 2021-06-28:
In what phase are you trying to read it?
1. Make sure to export it.
2. Note that modules only export values starting with their name, eg the base module variables start with ``${BASE_...}``.
Behavior coded here (related to how compgen works):
https://github.com/guysoft/CustomPiOS/blob/devel/src/config#L105

cooked, 2021-06-30:
@guysoft 

1) I'm setting up the workflow as follow:

[image]

[image]

then in the config file I'd like to use that variable to select only a subset of the modules available under src/modules

[image]
 
2) I believe I'm not in that scenario since the variable is used before entering the modules, am I wrong?

guysoft, 2021-06-30:
I think the best way is to add what you want to config.local. You can see how it can be used in this part of a github action:
https://github.com/guysoft/OctoPi/blob/devel/.github/workflows/build.yml#L41

davidzwa, 2026-01-17:
@cooked it looks like your question has been answered by the CustomPiOS maintainer. I will close this issue!