guysoft/CustomPiOS #134: Question: is it possible to pull a docker image from a local docker registry during start_chroot_script

jsimonrichard, 2021-07-02:
Would it be possible to pull a custom docker image from a docker registry on my own computer during the `start_chroot_script` (or through some other method)? I would like that image to be accessible after the custom distro is installed even if there is no access to my docker registry at that time.

A couple of issues come to mind, but I'm not sure how to solve them.
* What is my host machine's IP address relative to the chroot environment (is it just `localhost`?)
* Would docker be accessible during `start_chroot_script`?
* Can the docker daemon be started in the chroot environment?

Thanks, in advance, for your help.

guysoft, 2021-07-02:
Why would you want to do that?

jsimonrichard, 2021-07-02:
The product I'm developing will be installed on the lower levels of factory networks. I can't guarantee these devices will have access to the internet when they're installed.

guysoft, 2021-07-03:
You could try use something like this to ship the image as a tar. Then load it on boot:
https://stackoverflow.com/a/23938978/311268

jsimonrichard, 2021-07-06:
Yeah, that's probably the way to go.

Thanks for the help.

closed 2021-07-06 by jsimonrichard