guysoft/CustomPiOS #136: `<distro>-wpa-supplicant.txt` does not work if wifi details are entered with Raspberry Pi imager

cp2004, 2021-07-03:
This is a kind of theoretical issue, so I'm opening it for discussion.

If a user flashes an image using Raspberry Pi Imager they have the option to setup wifi in the UI there. Example https://octoprint.org/download/ - I recently changed since OctoPi is in RPi Imager now. This generates a script in `/boot/firstrun.sh` like this:

```bash
#!/bin/bash

set +e

cat >/etc/wpa_supplicant/wpa_supplicant.conf <<WPAEOF
country=GB
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
ap_scan=1

update_config=1
network={
	ssid="***********"
	psk="***********"
}

WPAEOF
chmod 600 /etc/wpa_supplicant/wpa_supplicant.conf
rfkill unblock wifi
for filename in /var/lib/systemd/rfkill/*:wlan ; do
  echo 0 > $filename
done
rm -f /boot/firstrun.sh
sed -i 's| systemd.run.*||g' /boot/cmdline.txt
exit 0
```

First, am I correct in thinking this will break the link between `/boot/octopi-wpa-supplicant.txt` and `/etc/wpa_supplicant/wpa_supplicant.conf` ([here](https://github.com/guysoft/CustomPiOS/blob/a1baeeca6fa17eaeecfae629d3e8759d2e3fb51f/src/modules/network/start_chroot_script#L41))?

Can you think of any way this could be made to work, so that we can use the more user-friendly method of downloading & installing OctoPi but still keeping `/boot/octopi-wpa-supplicant.txt` working? I'm wondering if it is only doable with some smart boot-time script, but that just seems overly complex to me.

guysoft, 2021-07-03:
1. Symlinks are not broken. It about be sent to the symlink path.
2. Strange that Rpi are automatically echoing GB as the country without asking the user after going out of their way to get users to set the right country
3. Has anyone actually tested it?

cp2004, 2021-07-03:
> Symlinks are not broken. It about be sent to the symlink path.

My brain was not in gear when I was thinking of this. Understood, seems it should still work then.

> Strange that Rpi are automatically echoing GB as the country without asking the user after going out of their way to get users to set the right country

I am in the UK - and have GB selected, so that's why that's there. Below the WiFi password box is a select country box. It pre-fills based on the device you are flashing from I believe.

Since you know more about this, I'm happy that it should still work as intended. I have tested that adding the WiFi config works, but this was (as stated) a theoretical thing that crossed my mind today. Thanks 👍

closed 2021-07-03 by cp2004