guysoft/CustomPiOS #137: Add a persistent /build_cache directory?

EternityForest, 2021-07-04:
Just a minor thing, but it could be a major help in some cases.  What if we had a directory in the workspace that was mounted at /build_cache, and persisted between builds?

This would let you download very large files within build scripts without wasting bandwidth.

I can make a PR or work on this myself if there is interest!

guysoft, 2021-07-04:
Could be done, the question is what are you trying to use it for?
In a way you already have this with ``BASE_APT_CACHE`` https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/base/config#L60

The code for that happens here:
https://github.com/guysoft/CustomPiOS/blob/devel/src/custompios#L115

EternityForest, 2021-11-17:
I'm currently using a forked version with just these lines added, and I use it for a lot of things, like automatically fetching things from git repos, and for downloading the .NET runtimes.  It's also useful for compiling things. There's no reason to recompile something when not much has changed, I can just do the whole compile right in the /build_cache.

``` bash
  mkdir -p $BASE_WORKSPACE/build_cache
  mkdir -p $BASE_MOUNT_PATH/build_cache
  mount --bind $BASE_WORKSPACE/build_cache $BASE_MOUNT_PATH/build_cache

```

guysoft, 2021-11-17:
ZynthianOS have a similar thing, what they do is place under the filesystem module what they have built and then unpack it in to the build.
I can see how a cache could be helpful, but thinking what kind of commands you would want to run in the build could help.
You could have seperate builds steps that use for example [``custompios_export` commands](https://github.com/guysoft/CustomPiOS/wiki/Common.sh-functions#custompios_export) and then move the results in to a filesystem section of a module.